Opentrons/opentrons #4230: MustHomeError being raised to client level

btmorr, 2019-10-15:
## overview

During labware calibration (and possibly at other times), a `MustHomeError` is being raised by the smoothie driver, and is not handled such that it interrupts RPC calls and is surfaced to users. This is both confusing, and also can make it impossible for the user to proceed through certain flows.

## steps to reproduce

Not entirely clear how to reproduce on other robots, but one of the robots in the lab is currently experiencing this issues as of today (15 Oct 2019).

## expected behavior

When smoothie raises a `MustHomeError` (or other internal operations signals), these should always be handled below the client level, and users should be able to recover and proceed through flows.

btmorr, 2019-10-15:
Run logs:
[serial-musthomeerror.log](https://github.com/Opentrons/opentrons/files/3730810/serial-musthomeerror.log)
[api-musthomeerror.log](https://github.com/Opentrons/opentrons/files/3730811/api-musthomeerror.log)

btmorr, 2019-10-15:
Protocols:

[Archive.zip](https://github.com/Opentrons/opentrons/files/3730815/Archive.zip)

Fabbe, 2020-03-28:
When simulating my protocol I get no errors on my laptop, on the OT2 I get some out of bounds errors (which I've been told to ignore). 
When executing the protocol I get a MustHomeError when I call instrument.pick_up_tip().
Any ideas?

SyntaxColoring, 2020-05-02:
Added the `investigate` label. I'm not sure if this is still happening on recent versions.

closed 2020-10-07 by asogluizzo