Opentrons/opentrons #4262: Labware slicing throws an error when providing a past-the-end stop index that's a string

SyntaxColoring, 2019-10-21:
# Overview

The following code should print the wells A9, A10, A11, A12.

```python
from opentrons import labware
plate = labware.load("corning_96_wellplate_360ul_flat", 1)

# Plate is 8x12.
print(plate.rows["A"]["9":"13"])
```

But it throws:

```
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/bin/opentrons_simulate", line 10, in <module>
    sys.exit(main())
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/opentrons/simulate.py", line 238, in main
    runlog = simulate(args.protocol, log_level=args.log_level)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/opentrons/simulate.py", line 165, in simulate
    exec(protocol.contents, {})
  File "sunny.py", line 5, in <module>
    print(plate.rows["A"]["9":"13"])
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/opentrons/legacy_api/containers/placeable.py", line 137, in __getitem__
    return self.get_children_from_slice(name)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/opentrons/legacy_api/containers/placeable.py", line 321, in get_children_from_slice
    s.start, self.get_index_from_name(s.stop), s.step)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/opentrons/legacy_api/containers/placeable.py", line 310, in get_index_from_name
    self.get_child_by_name(name))
ValueError: None is not in list
```

Apparently because "13" is not a valid column index. This leaves slices with string indices unable to address the full range of columns or rows.

# More details

* `plate.rows["A":"I"]` (where `"I"` is one past the last column, `"H"`) throws a similar error.
* `plate.rows["A"]["9":"12"]` works, except of course that it excludes the last well.
* `plate.rows["A"][8:12]` (with integer indices) works.
* `plate.rows["A"]["9":]` (without an explicit stop index) works.

Tested in APIv1. Untested in APIv2. (Does APIv2 even have slicing like this?)

bndo, 2020-01-06:
There is a workaround for this so we will update the docs and label this as a known issue.

mcous, 2021-12-10:
Closing since PAPIv1 was deprecated years ago

closed 2021-12-10 by mcous