Opentrons/opentrons #4356: [Labware Defs] Update Mag Deck Engage height values in labware definitions

mikec116, 2019-10-31:
## Context: 
1. All standard labware that is compatible with the mag deck contains an engage height value in its labware definition
2. We'd like to switch the meaning of "engage height = 0" from the magnet being located at the limit switch at the bottom of the mag deck to the plane where the labware meets the mag deck surface

## Proposed options (choose based on LOE for development): 
- Change all compatible labware defs to contain updated engage height values...
- OR add a new engage height key to the labware defs that can be used going forward

sfoster1, 2019-10-31:
We should go with option 2.

IanLondon, 2020-01-09:
End of 2019 dev team met and decided that this would be done by adding the offset in Python + PD, keeping the definitions themselves unchanged.

closed 2020-01-09 by IanLondon