Opentrons/opentrons #4360: Don't re-simulate protocol if file hasn't changed

pyeguy, 2019-10-31:
## overview

When loading a protocol that has successfully been simulated in the past or resetting after a completed run the loading of the protocol seems like it runs through a simulation step again which can be several minutes for complicated protocols.  

It seems like a checksum could get generated and cached for each protocol and if it hasn't changed the last simulation output (either success or error) could be used as the value instead of re-running the simulation. 

## current behavior

Slow load (assumed to be the simulation step) on any complex protocol initial load or reset after completed or canceled run. 

## steps to reproduce

Load a complex protocol. 

## expected behavior

Load quickly if the same (identical as validated by checksum) protocol file has been used before.

tzeitim, 2020-12-15:
I suffer from the same phenomenon. Having to wait between runs defies a bit the purpose of automation.

amitlissack, 2021-01-28:
This is an area we are focusing on right now. 

While it's not quite what you are looking for,  simulations in version 4.1 should be 40% of what they were before.

pyeguy, 2021-01-29:
glad to hear it @amitlissack . Sorry  Don't know the internals on this but seems like it could be pretty easy as keeping a cached set of md5's and bypassing the simulation if it's present. Since these are small text files could be as simple as:

```python
cache = set()
import hashlib
cache.add(
  hashlib.md5(open("<protocol_filename>",'r').read()).hexdigest()
)

...

if hashlib.md5(open("<protocol_filename>",'r').read()).hexdigest() not in cache:
   simulate
else:
  pass

```

sorry for armchair developing if this is obvious and/or harder to implement for other reasons

amitlissack, 2021-01-29:
Thanks for the suggestion, @pyeguy. 

Sadly, skipping simulation has radical implications. We have to perform a simulation to determine the equipment required by the protocol as well as the list of steps that occur in the protocol. That's how the Opentrons app populates its UI.

What we've been experimenting with is a fast analysis phase that extracts the metadata I mentioned above.

mcous, 2022-03-04:
Re-running your most recent protocol no longer requires re-analysis as of `v5.0.0`. In the upcoming `v5.1.0`, the robot will remember its last 20 protocols (see #8474)

closed 2022-03-04 by mcous