Opentrons/opentrons #4368: Bug: mentioned-but-unused modules cause protocol crash

rgatkinson, 2019-11-05:
## overview

I have a Python protocol that contains the lines:

```
temp_slot = 11
temp_module = modules.load('tempdeck', slot=temp_slot)
```

Depending on other settings in the protocol, this module may or may not be used; presently, it is not. Which means that the analysis carried out by the app doesn't require it (good): it's not listed in the manifest of required parts, doesn't participate in calibration, etc.

However, when I go to run the protocol, I get a vague error (see #4367), which is ultimately rooted in an error logged in c:/users/bob/appdata/opentrons/roaming/logs/combined.log:

```
{"level":"debug","message":"Trackable event","timestamp":"2019-11-05T01:15:38.502Z","label":"renderer","meta":{"event":{"name":"runFinish","properties":{"error":"AbsentModuleError [line 111]: no module of name tempdeck is currently connected","protocolAppName":"","protocolAppVersion":"","protocolAuthor":"0a116ad46b9d246ce10335f3bfd988e1456a728fb161e5a7385f9ae3a786d4c1","protocolName":"Pairwise Interaction: Dilute & Master & Plate","protocolSource":"","protocolText":"fb032ce97084c057fb6aaab02811f7fc6299745281d740147caf77ea13c0652f","protocolType":"python","robotApiServerVersion":"3.14.0","robotFF_calibrateToBottom":false,"robotFF_deckCalibrationDots":false,"robotFF_disableHomeOnBoot":false,"robotFF_disableLogAggregation":false,"robotFF_shortFixedTrash":false,"robotFF_useOldAspirationFunctions":false,"robotFF_useProtocolApi2":false,"robotLeftPipette":"p10_single_v1.5","robotRightPipette":"p50_single_v1.4","robotSmoothieVersion":"edge-8414642","runTime":5,"success":false}},"optedIn":true}}
```

Thus, even though the temp module is unused, and excluded from calibration, etc., if it's not present at run time, the protocol crashes.

## current behavior

Protocol crashes if contains mentioned-but-not-used module that is not presently connected.

## steps to reproduce

See above.

## expected behavior

Module should either be required (preventing run from going forward) or should not cause protocol to crash (absence is ignored).

mcous, 2019-11-05:
Thanks for the details @rgatkinson! This is related to #234. We'll look into what it would take to fix this behavior in the current protocol API.

In the next major version of the protocol API, expect that any module and labware load calls will require those items to be present, regardless of whether or not they are "used"

sfoster1, 2019-12-10:
This is fixed in 3.15.0 by https://github.com/Opentrons/opentrons/pull/4538 (you now must attach all pipettes and modules mentioned in your protocols)

closed 2019-12-10 by sfoster1