Opentrons/opentrons #4375: other: Architecture documentation is out of date

theosanderson, 2019-11-05:
## overview
Specifically, if I understand correctly, https://github.com/Opentrons/opentrons/blob/edge/architecture-and-planning/backend-architecture.md is out of date in referring to ResinOS (if I am correct in understanding that this has now been migrated away from).

More generally, as a tech-savvy user it has been quite difficult for me to get a sense of how the OT2 now fits together - I am on top of the API, and I realise that this is the most important thing, but it would be great for me to be able to understand exactly what is on the Raspberry Pi in my OT2 and where this is defined in the codebase. It took me a while to find the buildroot repo, which if I am correct defines a significant amount of this?

[I know things are in flux at the moment, and this is all very much intended as an observation rather than a complaint! I also realise that things are set up for devs and not me, and that's fair enough!]

theosanderson, 2019-11-06:
OK I finally found this important directory: https://github.com/Opentrons/buildroot/tree/opentrons-develop/board/opentrons/ot2 I think it would be cool if that was highlighted on the buildroot main README.md, and if the buildroot repo was described in the opentrons README.md.

sfoster1, 2019-11-07:
You're totally right - we've done a bad job keeping that file up to date, and in general documenting the software architecture of the robot. We'll work to fix that and tag the PRs to this ticket.

SyntaxColoring, 2020-06-04:
The specific file that @theosanderson mentioned was deleted as part of #4717.

I'm leaving this ticket open because I don't think that materially changes the issue. Whereas we had out-of-date architecture documentation before, we just don't have any architecture documentation now. 😅

SyntaxColoring, 2021-06-03:
I'm closing this because I think it's been addressed by #6245. I've just reviewed [OT2_ARCHITECTURE.md](https://github.com/Opentrons/opentrons/blob/edge/OT2_ARCHITECTURE.md) and confirmed it's still correct.

Of course, I don't think this is the end of the story for architecture and platform documentation. But now that OT2_ARCHITECTURE.md exists, we can work on it incrementally as more specific gaps are identified.

closed 2021-06-03 by SyntaxColoring