Opentrons/opentrons #4391: Grayed out buttons should explain why they're grayed out

SyntaxColoring, 2019-11-06:
# Overview

In the screenshot below, the **Home** and **Restart** buttons are grayed out because I'm not connected to the robot. However, there is nothing to indicate that that is the reason.

[image]

# Desired behavior

Any time a button is grayed out in the App, it should have a tooltip (or something) to teach the user what they have to do to un-gray it.

mcous, 2020-08-13:
App button audit for tooltips if disabled below. Hopefully this list is fairly exhaustive, but I definitely could've missed some in individual wizards. That being said, these are the big ones and I think what this ticket needs to be concerned with.

General

- [x] Main navigation bar
    - Well defined tooltips for disable and notification reasons

"Robot" page

- Connect button
    - [ ] Disabled if robot-server unhealthy
- Update robot button
    - [x] Disabled if update-server unhealthy
    - [ ] Disabled if robot up to date
- Calibrate deck button
    - [ ] Disabled if robot-server unhealthy
    - [ ] Disabled if RPC API not connected
- Home all axes button
    - [ ] Disabled if robot-server unhealthy
    - [ ] Disabled if RPC API not connected
- Restart robot button
    - [ ] Disabled if robot-server unhealthy
    - [ ] Disabled if RPC API not connected
- Lights toggle
    - [ ] Disabled if robot-server unhealthy
- Check calibration button
    - [ ] Disabled if robot-server unhealthy
    - [x] Disabled if RPC API not connected
    - [x] Disabled if deck calibration known to be bad
- Wi-Fi connect selector
    - [ ] Disabled if robot-server unhealthy
- Download logs button
    - [ ] Disabled if robot-server unhealthy
- "Factory reset" button
    - [ ] Disabled if robot-server unhealthy
- Advanced setting toggles
    - [ ] Disabled if robot-server unhealthy
- Update Robot OS from file
    - [ ] Disabled if robot-server unhealthy
    - Note: button disable logic is currently broken: #5429 

"Pipettes and Modules" page

- Change pipette button is never disabled (page not available if robot-server down)
- Pipette settings button is never disabled (button not rendered if robot does not support pipette settings)
- Module firmware update button
    - [x] Disabled if module up to date
- Temperature / Thermocycler deactivate button
    - [x] Disabled if RPC API not connected 

"Protocols" page

- Proceed to Calibrate button
    - [x] Disabled if incorrect equipment attached to robot (note: tooltip positioning is a little broken)

"Calibrate" page

- Pipette items in calibration list
    - [x] Disabled if no pipette specified for mount
- Tip Rack items
    - [ ] Disabled if tiprack has already been calibrated
- Labware item
    - [ ] Disabled if tipracks have not yet been calibrated

"Run" page

- Run button
    - [x] Disabled if modules are not connected
    - [x] Disabled if door is open
- Pause/resume button
    - [x] Resume disabled if door is open
- Cancel button
    - [ ] Disabled if cancel is in progress
    - [ ] Seems to have some other disable conditions but they're a little unclear
- Reset run button
    - [ ] Disabled if run reset is in progress

"More" page

- App update button
    - [ ] Disabled if up to date
    - [ ] Might have other disable conditions

SyntaxColoring, 2020-08-13:
Oh man, thanks a bunch for that audit.

I actually stumbled upon this ticket recently and lamented not having been more concrete when I wrote it. I guess I thought it was self-explanatory, at the time. 🤦

mcous, 2022-03-07:
Closing this issue since the app is moving away from disabled buttons in general, and that work will be better tracked in individual feature tickets

closed 2022-03-07 by mcous