Opentrons/opentrons #4397: question: Connecting to Smoothie without OT Linux

theosanderson, 2019-11-07:
I am trying to do some development beyond the scope of typical "protocols" and therefore want a high level of control over the machine that is connected to the Smoothieboard. (I have considered the various alternatives). It seems to me that I might be able to take an SD card containing a standard Raspbian distro, run `pip install opentrons` and have control over the robot. Unfortunately it hasn't been quite so simple and I wondered if anyone had any tips.

I get
`
>>> robot.connect()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/bare/lib/python3.6/site-packages/opentrons/legacy_api/robot/robot.py", line 519, in connect
    self._driver.connect(port=port)
  File "/home/pi/bare/lib/python3.6/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 515, in connect
    self._connect_to_port(port)
  File "/home/pi/bare/lib/python3.6/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 535, in _connect_to_port
    baudrate=self._config.serial_speed
  File "/home/pi/bare/lib/python3.6/site-packages/opentrons/drivers/serial_communication.py", line 126, in connect
    port = get_ports_by_name(device_name=device_name)[0]
IndexError: list index out of range
`

In my current configuration if I try to specify a port I get:
`>>> robot.connect("/dev/ttyAMA0")
Traceback (most recent call last):
  File "/home/pi/bare/lib/python3.6/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 1123, in _setup
    self._wait_for_ack()
  File "/home/pi/bare/lib/python3.6/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 907, in _wait_for_ack
    self._send_command('\r\n', timeout=SMOOTHIE_BOOT_TIMEOUT)
  File "/home/pi/bare/lib/python3.6/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 943, in _send_command
    return self._send_command_unsynchronized(command, timeout)
  File "/home/pi/bare/lib/python3.6/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 965, in _send_command_unsynchronized
    5.0, DEFAULT_COMMAND_RETRIES)
  File "/home/pi/bare/lib/python3.6/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 1043, in _write_with_retries
    raise serial_communication.SerialNoResponse()
opentrons.drivers.serial_communication.SerialNoResponse`

My `/boot/cmdline.txt` currently contains (thanks to trying various things):

`dwc_otg.lpm_enable=0 console=tty1 console=ttyAMA0,115200 root=PARTUUID=d59534e0-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash`

Does anyone have any instincts about what I might be missing? Any thoughts much appreciated.

theosanderson, 2019-11-07:
```
>>> list_ports.comports()[0].name
'ttyAMA0'
```

theosanderson, 2019-11-07:
OK I made some progress by adding `dtoverlay=pi3-miniuart-bt` to `/boot/config.txt`. Can now connect to robot and ask it to home which it does, but then I get a timeout waiting for a response from Smoothie. Also `robot.set_rail_lights_on()` has no effect. Trying disabling bt now.

theosanderson, 2019-11-07:
Woop - I think I've finally got it.

My final `/boot/cmdline.txt`:
`dwc_otg.lpm_enable=0 console=tty1 root=PARTUUID=d59534e0-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash` (no ref to AMA0)

Final section of `/boot/config.txt`:
```
[all]
#dtoverlay=vc4-fkms-v3d
start_x=1
gpu_mem=128
#dtoverlay=w1-gpio
enable_uart=1
dtoverlay=pi3-miniuart-bt
```

At this point I could connect with `robot.connect("/dev/ttyAMA0")` (note you have to specify the full path because normally an environmental variable contains this) but after calling the home command the Smoothie homed but then failed to respond, and commands for turning on the lights work. The problem was that the GPIO pins weren't initialized. Running gpio.initialize() did not fix this on its own. What was missing was running the blinkenlights script (https://github.com/Opentrons/buildroot/blob/opentrons-develop/board/opentrons/ot2/rootfs-overlay/usr/bin/ot-blinkenlights) that starts on startup on a normal OT2. This initializes a bunch of key pins.

After running that script everything seems to work! I think it would be awesome to refactor lines 17 to 29 of that script as gpio.enable_robot() or similar and then expose that within robot. Then the Python library becomes self-contained.

theosanderson, 2019-11-08:
Wrote this up: https://github.com/theosanderson/Advanced_OT2/tree/master/Raspbian_OT2

theosanderson, 2019-11-08:
Leaving this open because of the suggestion for simple refactoring - feel free to close it though.

theosanderson, 2019-11-08:
Related issue: APIv2 does not allow the port to be specified in `protocol = opentrons.execute.get_protocol_api()` so requires the environmental variable to be set. Given that the port parameter is still present downstream it would seem useful to surface this?

sfoster1, 2019-11-08:
Yeah, you're right. There are probably some slightly thorny issues around making sure that function is called only once, but nothing impossible. Feel free to open a PR - and thank you for this and other contributions!

PR #4416 (2019-12-12): chore: refactor startup sequence logic from blinkenlights to API
## overview

A number of pins need to be set at power-on before the robot can work correctly. Previously some of these were defined only in the [ot-blinkenlights](https://github.com/Opentrons/buildroot/blob/opentrons-develop/board/opentrons/ot2/rootfs-overlay/usr/bin/ot-blinkenlights) script in the buildroot repo. This was confusing to newcomers: in some sense it was the only logic needed for the robot to pipette not to be found in the opentrons repo. This PR places that logic in a new function, `robot_startup_sequence` in `gpio` [... cut]
closed 2019-12-12 by sfoster1