Opentrons/opentrons #4402: Errors in API log are not immediately flushed to persistent storage

SyntaxColoring, 2019-11-08:
# Overview

If an OT-2 is power-cycled after encountering an error, that error may not show up in API logs downloaded after the power cycle.

# Reproducing

## Steps
1. Upload and run raise_when_run.py, provided below. The app will give you a "Run encountered an error" message.
2. Download logs. Observe that the exception was logged.
2. Immediately power cycle the OT-2.
3. Download logs again. Observe that the exception is no longer there.

Tested on APIv1. Not tested on APIv2.

## raise_when_run.py

```python
from opentrons import robot

if not robot.is_simulating():
    raise Exception("Raisin' in the sun.")
```

# Expected behavior

If an error of any kind happens – in the API log, or in any other log – it should be logged with an immediate flush, so that power-cycling the OT-2 will not throw it away.

sfoster1, 2019-11-15:
In 3.15.0 these files will sync every 30 seconds instead of the default 5 minutes. I'm not sure I want to go farther (or can without significant changes to journald)

closed 2019-11-15 by sfoster1

SyntaxColoring, 2019-11-15:
@sfoster1 Hm. 30 seconds is indeed better than 5 minutes (and may prove to be sufficient), but I'm surprised that journald doesn't have a real solution for this.

Could we do something like `journalctl --flush` on every message of `LOG_ERR` or above?