Opentrons/opentrons #4403: docs: OT1 docs are not marked as being only for OT1

theosanderson, 2019-11-08:
## overview

If I perform a Google search (as many people will) for "opentrons pipette" I get as the top API result a page entitled "Liquid Handling — Opentrons API v2.0" which I would expect to be what I want to use for the OT2. In fact it is the OT1 API, but the only indication of this is in the URL: https://docs.opentrons.com/ot1/pipettes.html

## expected behavior

Put a big warning on all these pages that they are for the OT1 only.

DerAndere1, 2019-11-25:
@theosanderson and other users: I also stumbled across this confused google result. For people who are looking for the API v2 (BETA, may be subject to change, currently for the OT-2 robot), see this documentation: https://docs.opentrons.com/v2/new_protocol_api.html .

closed 2019-12-10 by sfoster1

sfoster1, 2019-12-11:
Note that even though this ticket is marked as closed, we probably won't deploy the docs change until the next release so we don't accidentally deploy documentation only valid in e.g. 3.16.0 and confuse everybody.

PR #4586 (2019-12-10): chore(docs): put a banner in ot1 docs
This lets you know that you're in the ot1 docs. Great!
It's the same style as the v1/v2 redirect banners.

Closes #4403
commit 5a23b6ef (2019-12-10): chore(docs): put a banner in ot1 docs (#4586)

This lets you know that you're in the ot1 docs. Great!
It's the same style as the v1/v2 redirect banners.

Closes #4403