Opentrons/opentrons #4422: App: Flip temperature module rendering in 3,6,9

b-cooper, 2019-11-11:
## overview

When a temperature module is placed in slots 3, 6, or 9, the Run app displays it as if the cords were on the left side. We should treat all tempdecks in slots 3, 6, 9 as flipped in the RobotWorkSpace component.

kinnari-opentrons, 2019-11-11:
[image]
Software version 3.14.0

bndo, 2019-12-16:
We're not going to migrate the designs from PD into this, but rather we can just flip it.

closed 2020-08-20 by b-cooper

kinnari-opentrons, 2020-08-20:
> Closed #4422 <https://github.com/Opentrons/opentrons/issues/4422> via #6383 <https://github.com/Opentrons/opentrons/pull/6383>.

So glad this is closed!

commit 92167e24 (2020-08-18): fix(app): flip modules on deck map when in slots 3, 6, or 9

This patches a visual bug that rendered modules in slots 3, 6, and 9 with their plugs towards the
center of the deck. This doesn't match the physical arrangement and occludes other information on in
the visualization. Any module in those right-most slots will now render in the correct orientation.

Closes #4422
PR #6383 (2020-08-20): fix(app): flip modules on deck map when in slots 3, 6, or 9
# Overview

This patches a visual bug that rendered modules in slots 3, 6, and 9 with their plugs towards the
center of the deck. This doesn't match the physical arrangement and occludes other information on in
the visualization. Any module in those right-most slots will now render in the correct orientation.

Closes #4422

# Changelog

- add an `extraTransform` prop to the `RobotCoordsForeignDiv` that appends to the svg transform if flipped.
- add logic to the `Module` component (only used in RA calibration tab deck map and protocol- [... cut]
commit d0347da7 (2020-08-20): fix(app): flip modules on deck map when in slots 3, 6, or 9 (#6383)

This patches a visual bug that rendered modules in slots 3, 6, and 9 with their plugs towards the
center of the deck. This doesn't match the physical arrangement and occludes other information in
the visualization. Any module in those right-most slots will now render in the correct orientation.

Closes #4422