Opentrons/opentrons #4426: Feat request: Bring back point labware

theosanderson, 2019-11-11:
#  overview

I'm sure you've given all this stuff a lot of thought, and I know user-friendliness is key, but I just wanted to share that I miss the "point" labware. There is nothing stopping me (I think) creating a 1x1 piece of labware width length,width, height of 0 (or 0.0001) which would in most ways be pretty similar. I guess I would need a different piece of labware for each "point", which is fair enough. The main limitation is that each point would use up a slot. I just tried tricking it into letting me use a "13th slot" with:
```
protocol.deck.data[13]=None
protocol.deck._positions[13] = protocol.deck._positions[12]
my_point = protocol.load_labware("my_special_point_thing",13)
```

But (reasonably) this didn't show up in the calibration.

## desired behavior

The ability to add a 'point' labware which doesn't get associated with any position on the deck, but still shows up on the left hand-side for calibration (or an optional parameter that would disable checking whether two bits of labware were in the same slot).

theosanderson, 2019-11-11:
Suggestions for other ways I can hack this also welcome.

theosanderson, 2019-11-12:
Well maybe https://github.com/Opentrons/opentrons/pull/4431 will solve my issues :)

mcous, 2019-11-12:
@theosanderson all liquid handling actions in PAPIv2 take a [`Location`](https://docs.opentrons.com/v2/new_protocol_api.html#opentrons.types.Location) as their location argument, so you could construct your own locations separate from the labware and slot system. These would not show up in the calibration list, because the calibration list is (meant to be) used for accounting for differences in the labware definition compared to real life due to manufacturing differences, slot springs, etc.

Otherwise, you could use the [Labware Creator](https://labware.opentrons.com/create/) (or manually write your own JSON definitions according to [Labware Schema](https://github.com/Opentrons/opentrons/blob/edge/shared-data/labware/schemas/2.json) and load the labware with [`ProtocolContext.load_labware_from_definition`](https://docs.opentrons.com/v2/new_protocol_api.html#opentrons.protocol_api.contexts.ProtocolContext.load_labware_from_definition). This could get a little tedious, but I suppose it's an option

theosanderson, 2019-11-12:
Thanks @mcous! Very useful to know that about Location.

Yeah I understand the rationale. It's just on the OT1 it was very easy to make a protocol that said:

- First bring the pipette tip to this exact point here
- Then bring it to that exact point there

This very basic feature has been lost and I think that's a shame and that there are ways it could be brought back without really compromising anything. But actually I think slot-sharing may fix everything for me. Will close this to reduce spam!

closed 2019-11-12 by theosanderson