Opentrons/opentrons #4430: opentrons_6_tuberack_falcon_50ml_conical not consistent with opentrons_10_tuberack_falcon_4x50ml_6x15ml_conical

rgatkinson, 2019-11-11:
# overview

In the definition of the tube rack ``opentrons\shared-data\labware\definitions\2\opentrons_10_tuberack_falcon_4x50ml_6x15ml_conical\1.json``, the 50ml Falcon tubes have the following coordinates:

```
    "A3": { "x": 71.38, "y": 60.25, }
    "B3": { "x": 71.38, "y": 25.25, }
    "A4": { "x": 106.38, "y": 60.25, }
    "B4": { "x": 106.38, "y": 25.25, }
```

The tube rack in ``opentrons\shared-data\labware\definitions\2\opentrons_6_tuberack_falcon_50ml_conical\1.json`` physically shares these same positions (the dimples on the labware bottom are re-used for example) while adding two more 50ml tubes. The positions are listed as follows:

```
    "B1": { "x": 35.5, "y": 25.24, },
    "A1": { "x": 35.5, "y": 60.24, },
    "B2": { "x": 70.5, "y": 25.24, },
    "A2": { "x": 70.5, "y": 60.24, },
    "B3": { "x": 105.5, "y": 25.24, },
    "A3": { "x": 105.5, "y": 60.24, }
```

The last four of these six wells should have indentical coordinates to those in the first rack, but they do not. In particular, the ``x`` coordinates are materially different (the ``y`` coordiantes also unexpectedly differ, but only by 10uM). Measurement with a micrometer seems to indicate that the 71.38 and 106.38 measurements are the correct ones, meaning that the second rack should be using ``x`` coordinates of 36.38, 71.38 and 106.38 instead of 35.5, 70.5, and 105.5.

## current behavior

Labware definitions are not mutally consistent.

## steps to reproduce
 
See above.

## expected behavior

Identical locations in different labware definitions should use identical coordinates.

rgatkinson, 2019-11-11:
This was unsettling to discover. Accurate positioning & dimensioning is the whole raison d'etre of the OT2, and I had an expectation that the positioning regarding the labware manfactured by Opentrons itself (which could be taken from the engineering drawing of the parts themselves) would be flawless. Clearly this is not the case.

Perhaps a broader review is warranted.

mcous, 2019-11-12:
Thanks for bringing this to our attention @rgatkinson. It does look like there's a slight, <1 mm difference in the X-positions across these parts. For this specific instance, the well spacing is still defined correctly, so the calibration system should take care of any problems that this definition deviance would cause.

From looking at the mechanical drawings, the part + assembly tolerance stackups are greater than this difference (hence the calibration system). That being said, you're right to assume the definitions should reflect the nominal dimensions, so a difference shouldn't happen. I'll be looking into all of these