Opentrons/opentrons #4433: Tip pick_up_distance should be a property of tip_racks not pipettes

theosanderson, 2019-11-12:
## overview

Specifically, at the moment I have to place the tip significantly above the top of the tips when calibrating my P200X8 gen 1 with OpenTrons P200 filter tips. A similar issue was raised on Twitter with me too:

https://twitter.com/pedrosaurio/status/1189606442740924418

More generally, the amount one needs to move down into a tip to pick it up is *a feature of the tip*, rather than a general property of the pipette model. Conceivably one could have two different tip racks with two different brands of tip on a run (one might be a special purification tip for example) and these might require different pick_up_distances. And generally, at present there is no way to specify the pick_up_distance in for example the Protocol Designer, so it would need to be a feature of the tip rack.

## current behavior

pick_up_distance is a property of the pipette and one has to fake-calibrate the tip_rack to take account of its pick_up distance.

## expected behavior

Tip_racks have a new labware parameter of pick_up_distance which says how far the pipette has to move down from being flush with the top of the tip to being firmly on the tip.

theosanderson, 2019-11-12:
Actually IMO the lower-effort and more robust fix for this is to revert to calibrating the pushed-down-onto-tip position of the pipette, rather than the flush-with-the-top-of-pipette position.

mcous, 2019-11-12:
The pickup process for the OT2 is fundamentally different than with the OT1. The pick-up distance is **not the same as the amount of tip overlap**. Tip overlap is specific to a given pipette / tiprack combination, but also isn't involved in a tip pickup.

The OT2 picks up tips by:

1. Attempting to move down in Z by `tipPickupDistance` mm from the top of the tip
2. Bottom out on the tip before the pipette actually moves `tipPickupDistance`, thereby skipping steps and losing its known Z position
    - This is to create a good seal by continuing to push into the tip even after the pipette bottoms out
3. Home the z-stage of the pipette to re-establish its position

Tip pickup distance should be read more like tip pickup "distance", in that it's never intended to actually move that far. If you're finding your pipette is unable to pick up tips reliably, does increasing the pickup "distance" in the pipette's settings page help?

[image]

theosanderson, 2019-11-12:
Thanks @mcous , that's extremely useful - it would be great to document the fact that it is meant to miss steps (apologies if I've missed that). I _think_ I've found on occasion that my robot has been crushing tips against the bottom of the rack by pushing too hard. Maybe I need to lower the current? Now that I understand the system I'm in a better position to troubleshoot this. @open3dgenome - what happened when you started your tips too low? Was it just that you were scared by how much it was pushing down?

I also have a suspicion that something weird is going on with this in APIv2 calibration as I was having a lot of problems with tipracks and getting Z-limit hit errors. I'll investigate further in due course.

open3dgenome, 2019-11-12:
Thanks for those gifs, @mcous (it has been a month of no real support via email and this took just a few hours, I really appreciate it). 

@theosanderson : In our case, the arm was ramming 3x (each time stronger) and picking up the tips. It looked awful. The problem was that after this point there was a terrible screeching noise and the Z-axis would then somehow stall and lose the relative coordinates of the axis, rendering the whole downstream steps meaningless (e.g. would not pick up tips on the next round or would aspirate 10 cm above the plates, etc).

I had prepared some [slides](https://www.dropbox.com/s/i2s4cbtyrp2ndxu/opentrons_report.pptx?dl=1) when I first contacted support via email. It will be much clearer there since there are videos and examples.

I finally detached the cover of the carriage arm to see whether I had damaged the unit. Fortunately I didn't and by removing the cover the sound stopped!!  Still, the problem of having to adjust the tip pickup somewhere around 7-8mm above the tips persists.

open3dgenome, 2019-11-12:
I would like to piggyback: we are having issues with the calibration of the Z-position of the magnetic module, just in case it is related (when I have time I will open a proper issue and link it back here).

For example, when calibrating the tip, I tell the robot that the plate is 15cm above (just as a control), this position gets ignored and the protocol uses always the same -perhaps- hardcoded Z-positions. This translates into bent tips when I use different plates for the magnetic unit.

mcous, 2019-11-12:
@open3dgenome we're tracking issues with the calibration of plates on magnetic modules in #4046. Feel free to sound off in that thread with your specific experience; more data is always helpful

theosanderson, 2019-11-12:
@open3dgenome , in case it wasn't clear the message on pipettes seems to be that:

-  the "ramming" seems to be normal behaviour (even though I agree it feels alarming)
- The Z-axis is then meant to re-find its coordinates again by homing (and I can see it doing that in your video (though I believe you when you say it did get lost).

I'll let you know how I get on after implementing this.

theosanderson, 2019-11-13:
OK - have just managed calibration on PAPIv2 without any issues ( except calibrate-to-bottom as elsewhere). The deck flexes quite a bit with P200x8 pickup but it works. I also remember getting a scary sound early on as with @open3dgenome and worrying that something was broken (although it cleared up sooner). I wonder if it's something about the pipette mount adjusting itself as it receives force from the bottom for the first time. Anyway, closing this issue - thanks a lot for the explanation.

PR #4435 (2019-11-26): docs: Document pick_up_tip hardware procedure
Expands docstring to explain how tip pick-up works in hardware terms.

Resolves #4433 (for now).
closed 2019-11-13 by theosanderson