Opentrons/opentrons #4451: Labware Creator: Check 3 Deck Crosses Before Labware Test

umbhau, 2019-11-14:
As a OT-2 user who just created a new labware definition, I would like to confirm my OT-2 has a good deck calibration before testing my labware, so that I do not conflate calibration problems with labware definition problems.

## Acceptance Criteria
- [ ] Update labware test protocol to check all three deck calibration crosses (slots 1, 3, 7) before labware check.

## Design
- Note this update should be released in conjunction with an update the the labware test documentation (@howisthisnamenottakenyet). Adding blocked tag until we're ready to go on documentation.

pantslakz, 2019-11-27:
@umbhau any ideas why this is blocked?

umbhau, 2019-12-03:
Just the note from above in the design section (making sure we also have updated documentation ready to go)

PR #4647 (2020-01-03): feat(labware-library): add decks 1 and 7 to labwareTestProtocol before labware test

## overview

add decks 1 and 7 to labwareTestProtocol before labware test

## changelog

closes #4451 
-->

## review requests

- [ ] test on robot
closed 2020-01-03 by jen-fong