Opentrons/opentrons #4457: PD: Nice-to-have: Auto-fill form fields with previously entered values

mikec116, 2019-11-15:
#User Story
As a user, I don't want to have to enter the same values over and over again when creating multiple protocol steps. 

# Acceptance Criteria
- [ ] If a user defines a magnetic module engage height earlier in the protocol, let’s preserve this number the next time the user tries to engage the magnetic module 
- [ ] If a user defines a tip position earlier in the protocol, let’s preserve this number the next time the user tries to define a tip position

pantslakz, 2019-11-15:
re tip position - should it preserve and replicate again for every step created after? or only steps referencing the same labware?

mikec116, 2019-11-15:
Removing the tip position feature

IanLondon, 2020-06-30:
Completed in PD Magnetic Module project

closed 2020-06-30 by IanLondon