Opentrons/opentrons #4460: Avoid reloading protocol unless absolutely necessary

rgatkinson, 2019-11-15:
## overview

Large python protocols can take 5 to 10 minutes to load, consuming operator time. Thus, reloading of protocols should be avoided unless absolutely necessary.

In particular, if a run is cancelled, and 'Reset Run' is clicked (per the UI prompt), the protcol currently gets reloaded. This seems unnecessary (though there might be deeper reasons).

## current behavior

Protocol is reloaded on 'Reset Run' after run is cancelled.

## steps to reproduce

See above.

## expected behavior

After cancelling a run, a reset should be possible w/o reloading the protocol.

mcous, 2019-11-18:
Thanks for the request. The simulation + run system is a particularly gnarly (technical term) part of our software, and will be rebuilt in the coming months as part of a larger push to improve the calibration and run experiences.

Unfortunately, due to how the current system chooses to share state and object instances between calibration an runs (which is bad!), implementing this functionality in the current system would be a little too risky for us.

You may not see any movement on this ticket for a while as we plan out some of the lower level requirements for this re-build, but it is our intention to allow for the storage and retrieval of calibration data for a given protocol

SyntaxColoring, 2021-06-03:
Closing this in favor of #4360.

closed 2021-06-03 by SyntaxColoring