Opentrons/opentrons #4461: Default / initial calibration for p10 tip is too low,

rgatkinson, 2019-11-15:
## overview

Write a protocol that uses a p10 (opentrons) tip. Do a factory reset that clears all labware and tip length data. Load protocol and calibrate pipetttes. Next, calibrate the tip: notice that the default / initial position at which the tip is calibrated is in fact below the top of the tips (i.e.: inside the tip barrel, if you're lucky, or crunching on the tip edge if you're not).

## current behavior

Default tip calibraiton for p10 tips is inside the tip barrel. This is unsafe (at worst).

## steps to reproduce

See above.

## expected behavior

With certainty, the default calibraiton position for _all_ tips and labware should be safe: i.e.: above the tip / labware in question.

SyntaxColoring, 2019-11-16:
I think I've observed something like this, but we should do some experimentation in-house to narrow this down. In particular:

* We should confirm that the problem seen here isn't from a bad deck calibration.
* If it is from a bad deck calibration, we should try to figure out where that bad deck calibration came from. I recently saw strange behavior after doing a factory deck calibration with a P10S, but didn't have time to investigate. It's possible that using a P10S to do deck calibration propagates wrong values, which could result in the observed behavior.

nusrat813, 2019-12-19:
P10M Observation - 

After resetting labware and tip length data, calibration of p10 Opentrons tiprack is off in x about 0.5mm. No issues with Z height. 

P10S Observation - 
After resetting labware and tip length data, calibration of p10 Opentrons tiprack is off in the X, Y and Z about 1.5 mm. If tip was not off in the X and Y, tip would be inside the barrel.

peradastra, 2021-04-01:
I think I am seeing this manifest as a result of an issue with the labware json

closed 2025-12-15 by sfoster1