Opentrons/opentrons #4462: [Investigate] Gen 1 pipettes pause for ~15 seconds after every tip ejection

mikec116, 2019-11-15:
## overview

Customers are reporting a long pause after pipettes drop tips which is resulting in significantly longer protocol run-times. 
This ticket is to track an investigation of this issue and define one or more potential solutions. 

Current Open Questions: 
- [ ] Is the delay caused by the pipette plunger's homing? 
- [ ] Is there an issue with the pipette's current? 
- [ ] Is this seen in Gen2s, or just Gen1 pipettes? 
- [ ] Is this seen in single-channel pipettes, or just multi-channels? 

## current behavior

After every tip ejection, the pipettes seem to freeze for ~15 seconds before progressing to the next step in the protocol. 

## steps to reproduce

1. Drop tips from a Gen 1 multi-channel pipette into the trash or a tip rack. 
2. Watch the pipette freeze before progressing to the next step in the protocol. 

## expected behavior

The pipette should resume movement immediately after tips are ejected.

johmperr, 2019-11-21:
I believe the drop_tip and return_tip homes afterwards. If you set that value to False then it should work. What i did was tell the tool when i wanted to drop the tip or return the tip instead of letting it do it via  a .transfer. Code was something like this p300.drop_tip([location],home_after=False) or p300.return_tip(home_after=False). Adding this solved the delay problem for me with Gen1 pipettes.

sfoster1, 2019-12-10:
So what's happening here is that P300, P50, and P10 Multi Gen 1 model 1.5 pipettes will go through their drop tip behavior twice. This behavior was added to make absolutely sure the tips get dropped - we were having customer reports that occasionally the tips wouldn't drop off the pipettes, which can ruin a protocol (of course, so can the added time!).

If you've never seen this issue, you can disable this behavior by editing the settings for your multichannel pipette. In the Opentrons App, click the robot tab and select your robot in the left pane. Click Pipettes and Modules, then click "Settings" under the multi pipette. Then, clear the checkbox next to "double drop tip":

[image]

Once you've done this for your multichannel pipettes, they will only do the drop tip behavior once. 

Sorry we couldn't get back to you sooner, but I hope this helps!

closed 2019-12-10 by sfoster1