Opentrons/opentrons #4465: USB connectivity should always be preferred over WIFI

rgatkinson, 2019-11-16:
## overview

I operate my OT2 from a laptop which is physically adjacent. I always plug in the USB cable. However, it's hit and miss as to whether when I scan for my bot in the Opentrons App whether it shows up as WIFI connectable or USB connectable. 

And even after being connected for a long time, this is still the case: I just completed a run lasting more than two hours for example, went to the app, did a 'refresh list' on the Robot pane, and my bot still comes up as WIFI connectable, not USB.

This is significant because the USB connection (when available) is more reliable and can have _significantly_ less latency than the WIFI connection. Using WIFI, I have, for example, experienced lag of nearly a second or so during calibration in response to my pressing an arrow key. And full-on dropouts occur from time-to-time when using WIFI (really annoying during calibration, as I then have to reload my protocol, which takes nearly ten minutes; less annoying once the protocol is running).

[image]

## current behavior

USB connectivity is rarely availalbe in the app, even when the robot is physically connected.

## steps to reproduce

See above.

## expected behavior

If the OT2 is physically connected by USB, the USB cable should always be used for connectivity.

rgatkinson, 2019-11-16:
The OT2 is definitely connected, yet is not discovered over USB by the app:

[image]

[image]

SyntaxColoring, 2019-11-16:
@rgatkinson, thanks for ticketing this.

The intent is definitely for the app to prefer USB when it's available. But the way USB connections work is a little temperamental. So, the app isn't preferring Wi-Fi – it's falling back to Wi-Fi when USB doesn't work, which is often.

Current workarounds include:
* Restarting the computer
* Restarting the network adapter
* Using Command Prompt to release the network adapter's IP and let it autoassign a new one for itself

This issue has proven difficult to track down because it only happens on some Windows 10 computers. If you could provide [some system information](https://support.opentrons.com/en/articles/3472478-retrieving-system-make-model), it would be very helpful in our investigation. You can email it to support@opentrons.com, if you'd prefer.

rgatkinson, 2019-11-16:
System info is as follows:

```
OS Name	Microsoft Windows 10 Pro
Version	10.0.18362 Build 18362
Other OS Description 	Not Available
OS Manufacturer	Microsoft Corporation
System Name	RGATKINSON
System Manufacturer	System manufacturer
System Model	System Product Name
System Type	x64-based PC
System SKU	SKU
Processor	Intel(R) Core(TM) i7-6700K CPU @ 4.00GHz, 4008 Mhz, 4 Core(s), 8 Logical Processor(s)
BIOS Version/Date	American Megatrends Inc. 3801, 2018-03-14
SMBIOS Version	3.0
Embedded Controller Version	255.255
BIOS Mode	UEFI
BaseBoard Manufacturer	ASUSTeK COMPUTER INC.
BaseBoard Product	MAXIMUS VIII EXTREME
BaseBoard Version	Rev 1.xx
Platform Role	Desktop
Secure Boot State	On
PCR7 Configuration	Binding Not Possible
Windows Directory	C:\WINDOWS
System Directory	C:\WINDOWS\system32
Boot Device	\Device\HarddiskVolume2
Locale	United States
Hardware Abstraction Layer	Version = "10.0.18362.387"
User Name	RGATKINSON\bob
Time Zone	Pacific Standard Time
Installed Physical Memory (RAM)	32.0 GB
Total Physical Memory	31.9 GB
Available Physical Memory	19.3 GB
Total Virtual Memory	36.7 GB
Available Virtual Memory	22.0 GB
Page File Space	4.75 GB
Page File	C:\pagefile.sys
Kernel DMA Protection	Off
Virtualization-based security	Not enabled
Device Encryption Support	Reasons for failed automatic device encryption: TPM is not usable, PCR7 binding is not supported, Hardware Security Test Interface failed and device is not Modern Standby, Un-allowed DMA capable bus/device(s) detected, TPM is not usable
Hyper-V - VM Monitor Mode Extensions	Yes
Hyper-V - Second Level Address Translation Extensions	Yes
Hyper-V - Virtualization Enabled in Firmware	No
Hyper-V - Data Execution Protection	Yes
```

rgatkinson, 2019-11-16:
System info from laptop is as follows:

```
OS Name	Microsoft Windows 10 Pro
Version	10.0.18362 Build 18362
Other OS Description 	Not Available
OS Manufacturer	Microsoft Corporation
System Name	RGATKINSON-2019
System Manufacturer	LENOVO
System Model	20KHCTO1WW
System Type	x64-based PC
System SKU	LENOVO_MT_20KH_BU_Think_FM_ThinkPad X1 Carbon 6th
Processor	Intel(R) Core(TM) i7-8550U CPU @ 1.80GHz, 1992 Mhz, 4 Core(s), 8 Logical Processor(s)
BIOS Version/Date	LENOVO N23ET65W (1.40 ), 2019-07-02
SMBIOS Version	3.0
Embedded Controller Version	1.15
BIOS Mode	UEFI
BaseBoard Manufacturer	LENOVO
BaseBoard Product	20KHCTO1WW
BaseBoard Version	SDK0J40697 WIN
Platform Role	Mobile
Secure Boot State	On
PCR7 Configuration	Elevation Required to View
Windows Directory	C:\Windows
System Directory	C:\Windows\system32
Boot Device	\Device\HarddiskVolume1
Locale	United States
Hardware Abstraction Layer	Version = "10.0.18362.387"
User Name	RGATKINSON-2019\bob
Time Zone	Pacific Standard Time
Installed Physical Memory (RAM)	16.0 GB
Total Physical Memory	15.9 GB
Available Physical Memory	6.80 GB
Total Virtual Memory	18.2 GB
Available Virtual Memory	6.35 GB
Page File Space	2.38 GB
Page File	C:\pagefile.sys
Kernel DMA Protection	Off
Virtualization-based security	Not enabled
Device Encryption Support	Elevation Required to View
Hyper-V - VM Monitor Mode Extensions	Yes
Hyper-V - Second Level Address Translation Extensions	Yes
Hyper-V - Virtualization Enabled in Firmware	Yes
Hyper-V - Data Execution Protection	Yes

```

rgatkinson, 2019-11-16:
FYI: If it's useful, I can arrange for y'all to be able to temporarily remote into this unit for debugging.

SyntaxColoring, 2019-11-16:
Thank you! I'll let you know if we decide to take you up on that offer.

mcous, 2019-11-18:
The screenshot posted _may_ be indicative of the robot failing to bring up its own link-local interface. We've observed this happen on one robot in house, but were not able to reproduce once we fixed it. Ping @sfoster1

rgatkinson, 2019-11-18:
Update, don't know if it's relevant, thought I'd relate just in case. Just rebooted both laptop and OT2 this Monday morning, and at the moment at least it's visible over USB:

[image]

rgatkinson, 2019-11-18:
Additional update:

* Rebooting OT2 (using 'reboot' command from ssh) caused loss of USB connectivity once reboot completed.

* Subsequent reboot of laptop restored same.

mcous, 2019-12-11:
Closing this ticket because:

- The app already prefers the USB-to-Ethernet IP address of a given robot if it can reach the robot at that IP address
- The lack of interface availability appears to be related to the computer itself in some way, in that rebooting the computer fixes the issue

The app doesn't interact with the interface / adapter directly; it just tries to hit IP addresses it knows about. If your computer isn't properly assigning a link-local address to the adapter and/or setting up routes, then the app isn't going to be able to reach the IP address of robot on the other end of the USB-to-Ethernet adapter and there's nothing the app can do about it.

There's definitely the possibility that there's also a problem on the robot side of things, but I'm unable to reproduce any sort of robot-side failure to self-assign a link-local address (via hard power cycle or via SSH `reboot`). Will continue to investigate and re-open and/or update this ticket accordingly

closed 2019-12-11 by mcous