Opentrons/opentrons #4481: Perf opportunity: special case 4x4 matrix inversion.

rgatkinson, 2019-11-20:
## overview

My protocol takes more than 4 1/2 minutes to load into the OT2, which gets really old for me, really quickly; I'm incented to find ways to reduce that execution time. As part of that effort, I ran my protocol in the Opentrons simulator under the Pycharm performance analyzer.

Surprisingly, the results indicated that a full 15.2% of my (simulated, desktop) time goes into inverting a 4x4 matrix in ``linalg.inv``:

[image]

This is using generic matrix inversion logic. Surely, it can only be an improvement to not use general purpose matrix inversion but instead take advantage that it's always 4x4. 

A general 4x4 matrix can be represented as:

[image]

If we defind ``d`` to be

[image]

the the inverse is:

[image]

That might be a little hard to see. In text form, d can be written as:

```
m14*m23*m32*m41 - m13*m24*m32*m41 - m14*m22*m33*m41 + m12*m24*m33*m41 + m13*m22*m34*m41 - m12*m23*m34*m41 - m14*m23*m31*m42 + m13*m24*m31*m42 + m14*m21*m33*m42 - m11*m24*m33*m42 - m13*m21*m34*m42 + m11*m23*m34*m42 + m14*m22*m31*m43 - 
   m12*m24*m31*m43 - m14*m21*m32*m43 + m11*m24*m32*m43 + m12*m21*m34*m43 - m11*m22*m34*m43 - m13*m22*m31*m44 + m12*m23*m31*m44 + m13*m21*m32*m44 - m11*m23*m32*m44 - m12*m21*m33*m44 + m11*m22*m33*m44
```

and the inverse, scaled by d, is:

```
-(m24*m33*m42) + m23*m34*m42 + m24*m32*m43 - m22*m34*m43 - m23*m32*m44 + m22*m33*m44	m14*m33*m42 - m13*m34*m42 - m14*m32*m43 + m12*m34*m43 + m13*m32*m44 - m12*m33*m44	-(m14*m23*m42) + m13*m24*m42 + m14*m22*m43 - m12*m24*m43 - m13*m22*m44 + m12*m23*m44	m14*m23*m32 - m13*m24*m32 - m14*m22*m33 + m12*m24*m33 + m13*m22*m34 - m12*m23*m34
m24*m33*m41 - m23*m34*m41 - m24*m31*m43 + m21*m34*m43 + m23*m31*m44 - m21*m33*m44	-(m14*m33*m41) + m13*m34*m41 + m14*m31*m43 - m11*m34*m43 - m13*m31*m44 + m11*m33*m44	m14*m23*m41 - m13*m24*m41 - m14*m21*m43 + m11*m24*m43 + m13*m21*m44 - m11*m23*m44	-(m14*m23*m31) + m13*m24*m31 + m14*m21*m33 - m11*m24*m33 - m13*m21*m34 + m11*m23*m34
-(m24*m32*m41) + m22*m34*m41 + m24*m31*m42 - m21*m34*m42 - m22*m31*m44 + m21*m32*m44	m14*m32*m41 - m12*m34*m41 - m14*m31*m42 + m11*m34*m42 + m12*m31*m44 - m11*m32*m44	-(m14*m22*m41) + m12*m24*m41 + m14*m21*m42 - m11*m24*m42 - m12*m21*m44 + m11*m22*m44	m14*m22*m31 - m12*m24*m31 - m14*m21*m32 + m11*m24*m32 + m12*m21*m34 - m11*m22*m34
m23*m32*m41 - m22*m33*m41 - m23*m31*m42 + m21*m33*m42 + m22*m31*m43 - m21*m32*m43	-(m13*m32*m41) + m12*m33*m41 + m13*m31*m42 - m11*m33*m42 - m12*m31*m43 + m11*m32*m43	m13*m22*m41 - m12*m23*m41 - m13*m21*m42 + m11*m23*m42 + m12*m21*m43 - m11*m22*m43	-(m13*m22*m31) + m12*m23*m31 + m13*m21*m32 - m11*m23*m32 - m12*m21*m33 + m11*m22*m33
```

Further performance gains might perhaps be obtained by looking at the actual matrix values that get inverted to see if simplifications emerge; this I have not done.

## current behavior

Loading a protocol spends > 15% of time in unnecessary general purpose matrix inversion.

## steps to reproduce

See above.

## expected behavior

Protocol loading should take significantly less time than it curently does.

sfoster1, 2019-11-25:
@rgatkinson Thanks for the report! I'm sorry it took so long to look at this, it's been a busy time over here with our upcoming release and its support for the thermocycler module and Python protocol api version 2.

The observed behavior certainly makes sense! It's worth testing the special-purpose implementation, but my instinct is that a pure python arithmetic implementation may be slower than the numpy implementation, which uses lapack-lite. 

An alternative fix is to check out version 2 of our Python protocol API, which is in beta in OT-2 software 3.14 and therefore should be available in the most recent opentrons wheel. V2 (which also includes a major refactor of the Python side of the motion control system) doesn't invert matrices in the hot path of commanding motion at all. Docs for V2 can be found [here](https://docs.opentrons.com/v2/) and an article on migrating protocols can be found [here](https://support.opentrons.com/en/articles/3425727-switching-your-protocols-from-api-version-1-to-version-2).

Unfortunately, the use of `inv` is extremely baked in to protocol api v1.

rgatkinson, 2019-11-25:
Thanks.

FWIW, here is a Python version of an inverter of 4x4 matrices:

```
def inv4x4(m):
    m00 = m[0, 0]; m01 = m[0, 1]; m02 = m[0, 2]; m03 = m[0, 3]
    m10 = m[1, 0]; m11 = m[1, 1]; m12 = m[1, 2]; m13 = m[1, 3]
    m20 = m[2, 0]; m21 = m[2, 1]; m22 = m[2, 2]; m23 = m[2, 3]
    m30 = m[3, 0]; m31 = m[3, 1]; m32 = m[3, 2]; m33 = m[3, 3]

    d = m03*m12*m21*m30 - m02*m13*m21*m30 - m03*m11*m22*m30 + m01*m13*m22*m30 + m02*m11*m23*m30 - m01*m12*m23*m30 - m03*m12*m20*m31 + \
        m02*m13*m20*m31 + m03*m10*m22*m31 - m00*m13*m22*m31 - m02*m10*m23*m31 + m00*m12*m23*m31 + m03*m11*m20*m32 - m01*m13*m20*m32 - \
        m03*m10*m21*m32 + m00*m13*m21*m32 + m01*m10*m23*m32 - m00*m11*m23*m32 - m02*m11*m20*m33 + m01*m12*m20*m33 + m02*m10*m21*m33 - \
        m00*m12*m21*m33 - m01*m10*m22*m33 + m00*m11*m22*m33

    result = [
        [(-(m13*m22*m31) + m12*m23*m31 + m13*m21*m32 - m11*m23*m32 - m12*m21*m33 + m11*m22*m33)/d,
            (m03*m22*m31 - m02*m23*m31 - m03*m21*m32 + m01*m23*m32 + m02*m21*m33 - m01*m22*m33)/d,
            (-(m03*m12*m31) + m02*m13*m31 + m03*m11*m32 - m01*m13*m32 - m02*m11*m33 + m01*m12*m33)/d,
            (m03*m12*m21 - m02*m13*m21 - m03*m11*m22 + m01*m13*m22 + m02*m11*m23 - m01*m12*m23)/d],
        [(m13*m22*m30 - m12*m23*m30 - m13*m20*m32 + m10*m23*m32 + m12*m20*m33 - m10*m22*m33)/d,
            (-(m03*m22*m30) + m02*m23*m30 + m03*m20*m32 - m00*m23*m32 - m02*m20*m33 + m00*m22*m33)/d,
            (m03*m12*m30 - m02*m13*m30 - m03*m10*m32 + m00*m13*m32 + m02*m10*m33 - m00*m12*m33)/d,
            (-(m03*m12*m20) + m02*m13*m20 + m03*m10*m22 - m00*m13*m22 - m02*m10*m23 + m00*m12*m23)/d],
        [(-(m13*m21*m30) + m11*m23*m30 + m13*m20*m31 - m10*m23*m31 - m11*m20*m33 + m10*m21*m33)/d,
            (m03*m21*m30 - m01*m23*m30 - m03*m20*m31 + m00*m23*m31 + m01*m20*m33 - m00*m21*m33)/d,
            (-(m03*m11*m30) + m01*m13*m30 + m03*m10*m31 - m00*m13*m31 - m01*m10*m33 + m00*m11*m33)/d,
            (m03*m11*m20 - m01*m13*m20 - m03*m10*m21 + m00*m13*m21 + m01*m10*m23 - m00*m11*m23)/d],
        [(m12*m21*m30 - m11*m22*m30 - m12*m20*m31 + m10*m22*m31 + m11*m20*m32 - m10*m21*m32)/d,
            (-(m02*m21*m30) + m01*m22*m30 + m02*m20*m31 - m00*m22*m31 - m01*m20*m32 + m00*m21*m32)/d,
            (m02*m11*m30 - m01*m12*m30 - m02*m10*m31 + m00*m12*m31 + m01*m10*m32 - m00*m11*m32)/d,
         (-(m02*m11*m20) + m01*m12*m20 + m02*m10*m21 - m00*m12*m21 - m01*m10*m22 + m00*m11*m22)/d]
    ]
    return np.array(result, copy=False)
```

rgatkinson, 2019-11-25:
Also: it's often the case that the matrix being inverted is the identity matrix; in such cases, the inversion can be avoided altogether. In particular, in ``pose_tracker.change_base``, if ``src is ROOT``, then ``up`` is empty, ``fold(up)`` is the identity matrix, and the inversion is unnecessary.

btmorr, 2019-11-27:
Thanks for the detailed information. We're about to embark on improving the whole movement/geometry system in the API server. Most of what we're planning to focus on is positional accuracy, but it will definitely be a good time to make performance improvements as well. I'll make sure this ticket gets included. Thanks again!

mcous, 2021-12-10:
No longer applicable due to the removal of PAPIv1 in 2020

closed 2021-12-10 by mcous