Opentrons/opentrons #4491: For debugging, provide a way to proceed despite not having the right pipettes or modules installed

SyntaxColoring, 2019-11-20:
For experimentation and debugging, Support occasionally wants to simulate, or even run, a protocol for which we are missing the correct pipettes or modules.

A recent change (for which I can't find the PR, sorry) completely prevents the user from running a protocol if they don't have all the right things installed. We would like something like a hidden feature flag to turn those errors back into warnings.

mcous, 2019-11-26:
I don't feel super game to implement this. The logic to block the user from these actions is:

- Really important, because doing stuff with the wrong pipettes attached has been a source of bad calibration data for pretty much the life of the software
- Already complicated, because of how little checking the server does and how much of that checking needs to and has been moved to the client

Adding additional logic for bailouts is a little too risky for my blood in the current sessions "architecture". Do you think you could maybe add more specific examples of diagnoses that you think would be (or have been in the past) useful via "incorrect" runs? I would hope we can find other path(s) forward to get you the necessary information.

WRT the ones you posted:

> Investigating errors as the app shows them, either from simulation or from an actual run

What is missing from or different in `opentrons_simulate` that doesn't satisfy this need?

> Investigating gross movement or calibration errors

I would assume that these sort of errors cannot be responsibly diagnosed without the proper equipment attached due to how the calibration data and properties of specific pipettes etc. get fed into movement logic. Is this not the right assumption in y'all's experience?

> Investigating gross movement or calibration errors

SyntaxColoring, 2019-12-02:
Fair points.

> What is missing from or different in opentrons_simulate that doesn't satisfy this need?

`opentrons_simulate` won't work for anything that has to do with the OT-2 operating system environment, as opposed to just the API. A big example is when someone follows the guide for using the camera in a Python protocol, but tries to save the file to a read-only directory. The App will show an opaque error as soon as the protocol starts to run, before any movement happens.

> I would assume that [gross movement] errors cannot be responsibly diagnosed without the proper equipment attached...

Yes, on second thought, you're right.

mcous, 2020-07-28:
@SyntaxColoring I kinda want to consider this closed by the hardware simlulation file capability added #5680. You can run a local dev server with whatever arbitrary equipment "attached", which I think gets at the best parts of this proposal. What do you think?

https://github.com/Opentrons/opentrons/tree/edge/robot-server/simulators

SyntaxColoring, 2020-07-28:
Yeah, that makes sense to me. 👍

closed 2020-07-28 by SyntaxColoring