Opentrons/opentrons #4504: Tip drop off issue with multichannel pipettes

Bozhie, 2019-11-22:
## overview

Using the multichannel pipettes (both p300m and p10m) the default drop tip behavior is not sufficient for the tips to release reliably.

## current behavior

When using multichannel pipettes with the OT2, we’ve had issues with tip ejection. Either the pipette tips are attached too tightly or the drop tip force is too weak to release the tips. We’ve increased the drop tip current to the max value (0.8mm) and the issue persists. Because the multichannel arm uses what looks like significant force to pick up tips, our work-around was to adjust the tip pick-up to 5-6mm above the tips. Others have had similar experiences #4433. Although adjusting the tip pick-up height helped, the tips would still sometimes fail to release in the middle of a protocol, causing jammed tips and shutting down the experiment.

It does not happen every time (every tip release or every protocol), but the issue has shown up often enough to make executing unsupervised protocols unreliable. 

## steps to reproduce

Robot server/app version 3.14.1
Using Gen1 pipettes
This has persisted for the past 2 software updates.
The p300m seems to jam more aggressively than the p10m, so we adjust tip pick up height to 5.5mm and 4.5mm above "flush", respectively. If the the tip pick up is calibrated as recommended ("flush" with tips) the issue occurs more frequently. We have not had this issue with any of the single channel pipettes. 

## expected behavior

Default tip pick up should reliably pick up tips without using so much force that they cannot be released.

theosanderson, 2019-11-23:
It could be worth trying sticking with the recommended 'flush' position but lowering the pick-up current.

yunhodavidkim, 2019-12-04:
Hi @Bozhie, it sounds like you might have an underlying hardware issue with your pipettes. I recommend that you contact our support team so that they can assist you in troubleshooting and, if necessary, provide you with replacement pipettes. You can contact our Support team by emailing them at support@opentrons.com or by messaging them through the Opentrons App.

mcous, 2019-12-04:
Closing this issue because it seems like an hardware issue. We'll make sure the support team is aware of this ticket in case an underlying software bug is uncovered or an additional pipette settings feature is need. In that case, we'll re-open this issue

closed 2019-12-04 by mcous