Opentrons/opentrons #4513: P300 multichannel throwing erroneous tip errors when using transfer() and distribute() methods

ColtonGarelli, 2019-11-26:
## overview

Adding and disposing of tips on p300 multichannel, regardless of that pipettes state (with or without tips) throws either a TipsNotAttached error or AssertionError tips already attached. To my knowledge, this error only occurs while using the transfer() and distribute() functions . Adding a remove_tips() command before the line throwing a tips attached error results in a TipsNotAttached error. The same occurs visa versa.

## current behavior

The transfer and distribute functions are behaving like they should automatically pick up tips upon call and an error is thrown if you attempt to pick up tips, whereas when you don't attempt to pick up tips the functions don't actually pick up the tips resulting in a downstream TipsNotAttached error.

## steps to reproduce

See attached screenshots for reproduction. 

## expected behavior

I would prefer to explicitly call pick_up_tip() and drop_tip() rather than implicit.

[image]
[image]

sfoster1, 2019-12-02:
@ColtonGarelli If you want to do your own tip management, you can pass `new_tip='never'` to the transfer command:
```
p300.transfer(100, source=reservoir.wells()[0], dest=plate.wells()[:3], new_tip='never')`. Documentation for this can be found [here](https://docs.opentrons.com/v2/new_complex_commands.html#new-tip).

let me know if that solves the problem!

mcous, 2019-12-03:
@ColtonGarelli as my colleague mentioned above, please use `new_tip='never'` to opt out of automatic tip handling.

In the latest released version of the software (3.14.1), the robot will attempt to do liquid handling actions even if it doesn't have tips attached, so **please be careful to always pick up and drop tips yourself if you opt out of automatic tip handling**. This bug (#4219) will be fixed in the next version of the software (3.15.0)

Regarding the specific issues you mentioned:

> The transfer and distribute functions are behaving like they should automatically pick up tips upon call and an error is thrown if you attempt to pick up tips

I am unable to reproduce this behavior on our latest development version of the robot software (the soon to be released 3.15). What version are you running on your robot? It's possible that you're hitting a bug that has been or will be fixed.

> the functions don't actually pick up the tips resulting in a downstream TipsNotAttached error

Ditto. In my testing, the automatic tip handling functions properly and pick up tips

closed 2019-12-03 by mcous

ColtonGarelli, 2019-12-06:
I guess I had been doing manual tip handling without encountering an error prior to here and I missed the new_tip='never' in docs. My bad. Thanks for the quick response @sfoster1 !