Opentrons/opentrons #4545: Cancel Function Cannot Override Pause

ahiuchingau, 2019-12-04:
## overview
When you happen to cancel a protocol in the App **exactly when** a `ProtocolContext.pause` happens in the protocol, the run would not be canceled successfully and stalled indefinitely.

```console
Dec 04 21:32:20 F09F8D80 opentrons-api[322]: Halting
Dec 04 21:32:20 F09F8D80 opentrons-api[322]: ProtocolContext.pause: {'msg': 'H1 of Corning 96 Well Plate 360 µL Flat on 1 Top'}
```
(This is not exactly the same behavior documented in the fixed issue #707)

## expected behavior
User should be able to cancel the run even when a pause is about to happen

mcous, 2019-12-04:
Talked in person with @Laura-Danielle and @ahiuchingau, and here's what we've got for a best guess:

Known:

- A paused protocol cannot be canceled because of a longstanding bug in how RPC session fundamentally works
- When a user clicks cancel, the app sends an RPC session.resume, followed immeidately by an RPC session.cancel

Speculation:

- When the session receives a session.resume, there is a potential race condition if the next command to be executed in the protocol is a pause
- If the protocol hits that pause command in between the execution of the RPC resume and RPC cancel, the protocol will be uncancelable

Given this, we're ice-boxing this ticket until RPC removal

amitlissack, 2021-01-28:
Another bug that may have been fixed by execution manager improvements?

b-cooper, 2021-01-28:
Can confirm that this bug is no longer reproducible on 4.1 (probably since earlier 3.x versions that incorporated execution manager).  Closing this ticket

closed 2021-01-28 by b-cooper