Opentrons/opentrons #4549: Paramiko ssh connection to OT2 using private key

ffffffrank, 2019-12-05:
## overview

## current behavior

## steps to reproduce

## expected behavior

btmorr, 2019-12-05:
In your Paramiko snipped, I don't see where you're adding the ssh key. I haven't used Paramiko before, but you'll need to supply it (probably via the `pkey` argument to `connect`)

theosanderson, 2019-12-06:
Maybe try try `ssh.connect(ip, username, key_filename=key_filename)`?

sfoster1, 2019-12-06:
Yeah I think @theosanderson is right. from the docs [here](http://docs.paramiko.org/en/2.6/api/client.html#paramiko.client.SSHClient.connect), `key_filename` is not the third positional argument - if you specify it with `key_filename=` it ought to work.

One quick thing to make sure of though - and you might already know this - is to make sure the saved private key is in the right format; if you generated it with ssh-keygen, it should be ok; but if you generated it with putty, you'll need to transition it to openssh format using a guide like [this](https://support.solarwinds.com/SuccessCenter/s/article/Converting-OpenSSH-and-PuTTY-style-keys)

ffffffrank, 2019-12-06:
> Maybe try try `ssh.connect(ip, username=username, key_filename=key_filename)`?

> Yeah I think @theosanderson is right. From the docs [here](http://docs.paramiko.org/en/2.6/api/client.html#paramiko.client.SSHClient.connect), `username` is not the second positional argument (it's the third) nor is `key_filename` the third (it's the seventh) - if you specify them with `username=` and `key_filename=` it ought to work.
> 
> One quick thing to make sure of though - and you might already know this - is to make sure the saved private key is in the right format; if you generated it with ssh-keygen, it should be ok; but if you generated it with putty, you'll need to transition it to openssh format using a guide like [this](https://support.solarwinds.com/SuccessCenter/s/article/Converting-OpenSSH-and-PuTTY-style-keys)

Hi,
Thanks for the replies.

Here is what I tried:
I generated the key pairs using: ssh-keygen
I uploaded the public key to the robot using:
@{key = Get-Content ot2_ssh_key.pub | Out-String} | ConvertTo-Json | Invoke-WebRequest -Method Post -ContentType 'application/json' -Uri ROBOT_IP:31950/server/ssh_keys

I can access the robot by using in cmd windows:
ssh -i key_file_path -o UserKnownHostsFile=NUL -o StrictHostKeyChecking=no root@IP

And I can execute command like: cd /.,  ls, it returns the files under /.
everything is fine. I can even control the robot moving.

But for what I tried in paramiko python:
here is the code:
import paramiko
ip = ip from opentron app
usr_name='root'
key filename = r'C:\.../.ssh/id_rsa'
ssh_client = paramiko.SSHClient()
ssh_client.set_missing_host_key_policy(paramiko.AutoAddPolicy())
ssh_client.connect(ip,username=usr_name, key_filename=keyfilename)

The wired thing is, when I run
stdin, stdout, stderr = ssh_client.exec_command('hostname')
print (stdout.read())

I got the robot serial number: OT2C***
That shows the ssh is indeed working.

But when I run:
stdin, stdout, stderr = ssh_client.exec_command('cd /.')
stdin, stdout, stderr = ssh_client.exec_command('ls')
print (stdout.read())
I got nothing as comparison to the returning of the files under / that I mentioned above when I perform the operation in bash in robot.

Also,
stdin, stdout, stderr = ssh_client.exec_command('bash -lc "python3 /python_script.py function"'))
nothing happened.
whereas if I ran the "python3 /python_script.py function" in robot bash directly, the robot is moving as I expected.

I appreciate your help, and thanks for spending time reading.

theosanderson, 2019-12-06:
> stdin, stdout, stderr = ssh_client.exec_command('cd /.')
> stdin, stdout, stderr = ssh_client.exec_command('ls')
> print (stdout.read())
Your `cd` will not be persisted to the `ls` command I think. `stdin, stdout, stderr = ssh_client.exec_command('ls /')` should work.

>stdin, stdout, stderr = ssh_client.exec_command('bash -lc "python3 /python_script.py function"'))
Did you try simplifying this to `stdin, stdout, stderr = ssh_client.exec_command('python3 /python_script.py function'))`? I'm not sure if there even is a bash installed on the OT2.

mcous, 2019-12-06:
Thanks @theosanderson! You're right; `bash` is not installed on the OT-2. Those insights about `exec_command` also appear to be correct based on my reading of the paramiko docs.

@ffffffrank I'm going to close this issue because there doesn't appear to be any underlying issue with the OT-2's SSH server. Feel free to continue to use this thread to ask questions about proper paramiko usage and we'll continue to answer them to the best of our knowledge!

closed 2019-12-06 by mcous

m89ozturk, 2022-06-10:
Hi Everyone,
I am trying to build an ssh connection to the OT-2 using paramiko but I receive the following error.
Any help is greatly appreciated.

CODE:
ssh = paramiko.SSHClient()
ssh.load_host_keys('.ssh/known_hosts')
ssh.set_missing_host_key_policy(paramiko.AutoAddPolicy())
SSHkey = paramiko.RSAKey.from_private_key_file("ot2_ssh_key", password="XYZ")
ssh.connect(hostname='169....', username='root', passphrase='XYZ', pkey=SSHkey)

ERROR:
File ".\client.py", line 435, in connect
    self._auth(
  File ".\client.py", line 766, in _auth
    raise saved_exception
  File ".\client.py", line 666, in _auth
    self._transport.auth_publickey(username, pkey)
  File ".\transport.py", line 1635, in auth_publickey
    return self.auth_handler.wait_for_response(my_event)
  File ".\auth_handler.py", line 259, in wait_for_response
    raise e
paramiko.ssh_exception.AuthenticationException: Authentication failed.