Opentrons/opentrons #4563: PD: Warn users when exporting a protocol that includes an unused pipette or module

mikec116, 2019-12-06:
## User Story
As an PD user, I should be warned (but not blocked) from exporting a JSON protocol that specifies hardware (pipettes + modules) that isn't actually used in the protocol. 

## Context
Protocols like this will simulate/load but won't run in APIv2 (unless the specified hardware is actually attached to the robot). 

Users will arrive at the File Info page - and it will say "pipette missing" or "module missing" - and until those pipettes and modules are detected on robot - they will not be able to perform the run.

Our PD warning might direct users to either remove the pipette/module from protocol before exporting, or ensure the pipette/module are attached to robot in order to successfully run protocol.

## UI Design
- TBD

## Acceptance Criteria
- [ ] User is warned when exporting a protocol with a pipette that is specified but not used
- [ ] User is warned when exporting a protocol with a module that is specified but not used
- [ ] User is warned when exporting a protocol with more than one module that is specified but not used
- [ ] User is warned when exporting a protocol with a pipette and a module that is specified but not used

howisthisnamenottakenyet, 2020-01-22:
@pantslakz @mikec116 @umbhau Would appreciate if one of you could peek at this before EOD and approve/give feedback on copy.

pantslakz, 2020-01-22:
Case 1
Title: Unused pipette
Body: The {left/right} {pipette type} **pipette specified** in your protocol is *currently* not used in any step. **In order to run this protocol successfully you will need to attach this pipette to your robot.** If you don't intend to use the pipette, **please** consider removing it from your protocol.
Cancel button: Cancel
Continue button: Export anyway

Case 2
Title: Unused module
Body: The {module} **specified** in your protocol is **currently** not used in any step.  **In order to run this protocol successfully you will need to power up and connect the module to your robot.** If you don't intend to use the module, **please** consider removing it from your protocol.
Cancel button: Cancel
Continue button: Export anyway

PR #5046 (2020-02-21): feat(protocol-designer): warn user when exporting a file with an unused pipette and module
## overview
Display warnings when there is an unused pipette or module when exporting a protocol.
closes #4563

## changelog
Add various warnings to alert i18n
Add new selector to grab pipette data from deck

## review requests
http://sandbox.designer.opentrons.com/pd_unused-module-or-pipette/
Verify that the warnings match the warnings from [here](https://github.com/Opentrons/opentrons/issues/4563)

Case 1 - Unused pipette
create a new protocol with a left and right pipette
create a transfer step that u [... cut]
commit 97866c6e (2020-02-21): feat(protocol-designer): warn user when exporting a file with an unused pipette and module (#5046)

* feat(protocol-designer): add warnings when user exports a file with an unused pipette or module

closes #4563

* add more tests

* fix flow typings

* make inexact types exact object types

* suggested change to merge the 2 getUnusedEntities fns

* clean up test file
closed 2020-02-21 by jen-fong