Opentrons/opentrons #4587: Default calibration height for tiprack-200ul is too low for a physical opentrons-96-tiprack-300ul

SyntaxColoring, 2019-12-11:
# Summary

The `tiprack-200ul` definition is purported to be compatible with modern Opentrons 300 µL tip racks. However, its default calibration height is too low in this case. It will put the pipette nozzle slightly inside the pipette tips, which is likely to cause a crash.

This impacts people's ability to run any legacy protocol with an out-of-the-box OT-2 or after a labware calibration factory reset. Notably our Labware Library protocols still use the `tiprack-200ul` definition.

# Steps to reproduce

1. Factory reset labware calibration.
2. Upload a protocol with a `tiprack-200ul`.
3. Load the deck with an Opentrons 300 µL tip rack.
4. Do labware calibration. Note that the initial calibration position will have the pipette nozzle inside the pipette tips, instead of at or above its top.

The default position of `opentrons_96_tiprack_300ul` looks more reasonable.

mcous, 2019-12-11:
> The tiprack-200ul definition is purported to be compatible with modern Opentrons 300 µL tip racks

@SyntaxColoring who told you this? I don't think it's true. Prior to `opentrons_96_tiprack_300ul` existing, we (SW + API docs) recommended users use the v1 definition `opentrons-tiprack-300ul`. If there are protocols on PL for OT-2 that reference the super-old `tiprack-200ul` (which, IIRC, is possibly a TipOne or GEB tiprack with a 3D printed adapter) then that's a problem in PL

SyntaxColoring, 2019-12-11:
Any other Opentrons folks, please see the Slack #support thread instead of replying here. 👍

PR #4619 (2019-12-18): docs(api): remove tiprack-200ul from v1 docs
## overview
This PR replaces all very ancient `tiprack-200ul` definitions and the not-as-ancient-but-still-out-of-date `opentrons-tiprack-300ul` to `opentrons_96_tiprack_300ul`.

closes #4587

## review request
make sure the old definitions are no longer being used in:
http://sandbox.docs.opentrons.com/docs_remove-old-tiprack-defs/v1/
commit f5541153 (2019-12-18): docs(api): remove tiprack-200ul from v1 docs (#4619)

closes #4587
closed 2019-12-18 by ahiuchingau