Opentrons/opentrons #4590: 3.15.0 or 3.15.1 may have changed tip drop behavior

SyntaxColoring, 2019-12-11:
# Summary
Multiple customers have reported tip drops becoming less reliable after upgrading to 3.15.0 or 3.15.1. 

# What to collect

We don't currently have full debugging files from any customer. The relevant things to collect are:

* api.log and serial.log, downloaded through the Opentrons App.
* Screenshots of the pipette settings shown in the Opentrons App.
* The pipette settings files stored on the OT-2. An amalgamated file containing this information can be created in the Jupyter notebook directory by running this command in a Jupyter terminal: `for f in /data/pipettes/*; do (echo $f; cat $f; echo) >> /var/lib/jupyter/notebooks/pipette_settings.txt; done`
* The serial numbers on the barcodes on the attached pipettes.
* Whether or not downgrading to 3.14.1 helps.

# Workarounds

Downgrading to 3.14.1 has solved the problem for 2 customers.

ffffffrank, 2019-12-12:
3.14.1, pipette drop_tip is not working either...

gonebald, 2019-12-12:
I can confirm the same issue in v.3.15.1, here are the debug files as requested.

Apparently, tip dropping seems to work normally when calibrating the labware but failure occurs when performing the run and dropping tips into the trash bin. This also seems to be random, i.e. the first drop_tip sometimes works well but then fails at the second or third tip.

When failure happens, the tip ejector moves only slightly - changing e.g. plunger position or drop tip current seems to make no difference.

I have not tested firmware downgrade, please provide instructions and source file(s) for that.

Included files:

1. api_part.log and serial_part.log - log excerpts for one run including calibration; first drop_tip (line 43) works, second drop_tip (line 62) fails
2. left-P50.png and right-P300.png - pipette settings screenshots
3. pipette_settings.txt - exported using the given command
4. robot_configs.py - current modified settings with reduced speeds and accelerations
5. test_protocol.py and test_data.txt - custom protocol and data for run replication.

[OT2_drop_tip_bug.zip](https://github.com/Opentrons/opentrons/files/3957137/OT2_drop_tip_bug.zip)

gonebald, 2019-12-12:
Just remembered, the issue first occurred for v.3.14.1 (possibly even earlier version, not sure), updated to v.3.15.1 for that reason.

Laura-Danielle, 2019-12-12:
Hi @gonebald and @ffffffrank, are you both experiencing these issues with multi-channel pipettes only?

ffffffrank, 2019-12-12:
> Hi @gonebald and @ffffffrank, are you both experiencing these issues with multi-channel pipettes only?

No, mine is single 1000uL pipette

gonebald, 2019-12-12:
likewise, single channel P50.

gonebald, 2019-12-12:
also, haven't noticed this issue with the P300 in the right slot (P50 is in the left slot), will try to reproduce and also switch pipettes (left-right).

ffffffrank, 2019-12-12:
@gonebald mine doesn't drop tip on either side:(

SyntaxColoring, 2019-12-12:
@gonebald @ffffffrank Thank you both for the information.

I want to make sure we're all talking about the same issue. Tip drop problems can happen for a few different reasons, including calibration problems and defective hardware.

The scope of this issue is that software is causing the action of the pipette during tip drop to be wrong. To confirm if this is the case, during a tip drop, watch:

1. To see how many times the ejector comes down.
2. To see how far the ejector comes down.

It's best if you can provide a video. You can send it to support@opentrons.com. Just mention your GitHub username, and that Max told you to send the video.

Thanks for your help!

ffffffrank, 2019-12-13:
@SyntaxColoring Hi Max, for my 1000uL pipet, the ejector doesn't move at all when I command drop_tip. What should I do?

gonebald, 2019-12-14:
here's a quick follow up, i recorded a video of both successful and failed tip drops: [OT2_drop_tip_failure_VIDEO.zip](https://github.com/Opentrons/opentrons/files/3964177/OT2_drop_tip_failure_VIDEO.zip)

interestingly, the first run went well with no failed tip drops indicating that this is not a hardware issue, but on the repeated run of the same protocol immediately after the successful run there was drop tip failure on the fourth tip. 

once the failure occurs, all subsequent tip drops fail.

after the second failed tip drop the OT app reports a smoothie error in line 115 of the protocol.

[image]

after restarting the robot and the app the same behavior repeats, i.e. the first few tip drops work and then failure occurs.

i ran the same protocol with the p300 pipette in the right slot and that seems to work normally.

gonebald, 2019-12-14:
smoothie error reported above occurs during the next pipette transfer after a failed tip drop.

i haven't yet tried to switch pipettes or downgrade the firmware.

SyntaxColoring, 2019-12-14:
@gonebald Thanks for that video. Having run that test, would you mind now uploading the **api.log** and **serial.log** files? You can get them from **Advanced Settings > Download Logs > Download**.

gonebald, 2019-12-16:
here are the requested logs.

the first corresponds to a successful run:

[SUCCESS-logs.zip](https://github.com/Opentrons/opentrons/files/3968685/SUCCESS-logs.zip)

the second corresponds to the failed run as shown in the uploaded video:

[FAILED-log.zip](https://github.com/Opentrons/opentrons/files/3968686/FAILED-log.zip)

both runs were performed using the same protocol and hardware.

the protocol and settings have been uploaded previously:

https://github.com/Opentrons/opentrons/issues/4590#issuecomment-565114058

closed 2019-12-16 by sfoster1

gonebald, 2019-12-17:
please advise how to fix the issue with updated firmware, no stable updates are available as of this morning.

mcous, 2019-12-17:
@gonebald we appreciate your patience! This issue tracker is oriented towards our codebase, so issues will be closed when the code is complete.

We are working on preparing this for release right now, which takes some time as we run the release through our QA process. When the update is released and available, we'll post a note in this thread and your app will notify you of the available update

gonebald, 2019-12-17:
ok, this is as expected, thanks for the follow-up.

gonebald, 2019-12-20:
updated with the firmware 3.15.2, the problem persists.

i have attached log files corresponding to two runs - first run is successfull (all tips ejected), second run is unsuccesfull (last tip not ejected).

please advise what to do next.

[logs.zip](https://github.com/Opentrons/opentrons/files/3988660/logs.zip)

SyntaxColoring, 2019-12-20:
Hi @gonebald,

Based on that, and your mention earlier that this actually started in 3.14.1 or before, I think you're experiencing a separate issue. Shoot an email over to support@opentrons.com and we'll do a separate investigation. 👍

PR #4611 (2019-12-16): fix(api): make sure the robot is connected before querying instrs
Since the robot singleton is now not the thing the rest of the api runs on, it
may not be connected the first time the system tries to detect attached pipettes
before a run. This means that requested pipettes get built as v1, which means
that their settings stand a good chance of being wrong - and that can lead to
failed drop tips.

Closes #4590

## Testing

Upload v1 protocols under the following circumstances and check the app redux state path /robot/session/pipettesByMount/{mount}/name. The name should usually (with some note [... cut]
commit 286f1ce6 (2019-12-16): fix(api): make sure the robot is connected before querying instrs (#4611)

Since the robot singleton is now not the thing the rest of the api runs on, it
may not be connected the first time the system tries to detect attached pipettes
before a run. This means that requested pipettes get built as v1, which means
that their settings stand a good chance of being wrong - and that can lead to
failed drop tips.

Closes #4590