Opentrons/opentrons #4598: `make install` should be named something else

SyntaxColoring, 2019-12-13:
# Overview

Our build system pervasively uses `make install` to mean something like "download and locally install the build dependencies."

This, unfortunately, conflicts with the conventional meaning of `make install`, which is "install the program we just built into the standard system directories." Or, as the [GNU Make manual](https://www.gnu.org/software/make/manual/html_node/Standard-Targets.html#Standard-Targets) puts it, to "compile the program and copy the executables, libraries, and so on to the file names where they should reside for actual use."

For the sake of reducing surprises, we should rename `make install` to `make dev-deps` or something.

# Considerations

- We should agree on a good new name. I'm not aware of any conventional ones.
- All instances of the `install` target will have to be updated.
- Readmes and other documentation will have to be updated.
- There should be a deprecation period in case there's a hidden CI script somewhere that's using `make install`.

# Work

If there are no objections in principle, I can give it a shot, as a vehicle for familiarizing myself with the breadth of the build system.

mcous, 2019-12-13:
Yeah, this one was a complete brain fart by me. I had `npm install` in my head when I set them up, despite the fact that I read and tried to follow the GNU conventions.

In Opentrons/opentrons-modules#50 we ended up going with `setup` and `teardown`. I don't have strong feelings on the task name itself, but any consistency is good

SyntaxColoring, 2019-12-17:
Cool.

@mcous Could you clarify the scope of `make install` as it currently exists? I see several kinds of dependencies:

1. Actual code dependencies, like libraries.
2. Development dependencies, like `pytest`.
3. Incidental dependencies just for the build system, like `shx`.

Does `make install` cover all of these? 

I ask because there might be a more precise name we can use. (Otherwise, `make setup` sounds good.)

mcous, 2019-12-17:
@SyntaxColoring it covers all three of those. Everything you need to build, test, and run any project in the monorepo.

Ignoring the task names, the tasks are configured in such a way to allow the setup steps for JS and Python to be run in parallel via `make install -j`. Please avoid touching any task structure for this reason

PR #5823 (2020-06-08): build(make): name targets "setup"/"teardown" instead of "install"/"uninstall"
## overview

This renames, repository-wide:

* `make install` ➡️ `make setup`
  * `make install-js` ➡️ `make setup-js`
  * `make install-py` ➡️ `make setup-py`
* `make uninstall` ➡️ `make teardown`

To match mainstream convention, and to be consistent with [Opentrons/opentrons-modules](https://github.com/Opentrons/opentrons-modules).

Closes #4598.

## changelog

* **Breaking: Renamed build targets, as described above.**
* Made the relevant adjustments to CI scripts.
* Updated all mentions of the target names in read [... cut]
closed 2020-06-08 by Kadee80