Opentrons/opentrons #4609: SynchronousAdapter.name must return a string, not func

rgatkinson, 2019-12-13:
## overview

``opentrons.hardware_control.adapters.SynchronousAdapter`` inherits from ``threading.Thread``. Addtionally, conceptually, it also sort-of inherits from ``opentrons.hardware_control.API`` through it's own delegation mechanism.

Both ``Thread`` and ``API`` have a ``name`` attribute.

``Thread.name`` returns a string, and the PyCharm debugger relies on this.

However, ``SynchronousAdapter.name`` returns ``API.name``, which returns (e.g.) ``TempDeck.name`` which results in an unbound method, not a string. This crashes PyCharm, and makes the OT2 protocol undebuggable, which is a bit of a big deal. 

A simple fix is to add 

```
        if attr_name == 'name':
            return super(SynchronousAdapter, self).name
```

at the top of ``SynchronousAdapter.__getattribute__`` (I've tested same with some judicious hacks).

## current behavior

See above.

## steps to reproduce

See above.

## expected behavior

All OT2 protocols should be debuggable in the simulator.

rgatkinson, 2019-12-14:
A different, perhaps more elegant, fix would be change from SynchronousAdapter 'is-a' Thread to SynchronousAdapter 'has-a' thread, which would intrinsically avoid this conflict and any others that might be lurking.

rgatkinson, 2019-12-14:
The following is a protocol which is not debuggable because of this bug:

```

from opentrons.protocol_api import ProtocolContext

metadata = {
    'protocolName': 'Test Test Test',
    'author': 'Robert Atkinson <bob@theatkinsons.org>',
    'apiLevel': '2.0'
}

def run(protocol_context: ProtocolContext):
    temp_module = protocol_context.load_module('tempdeck', location=11)

```

mcous, 2019-12-16:
After some digging, it looks like this class is not being a well-behaved sub-class of `threading.Thread`. We'll keep an eye on this and likely try to move away from this multiple-inheritance thing that's going on

sfoster1, 2020-03-02:
After #4870 SynchronousAdapter doesn't inherit from Thread anymore

closed 2020-03-02 by sfoster1