Opentrons/opentrons #4640: APIv2: pipette moves between mix() repetitions

SyntaxColoring, 2019-12-18:
# Current behavior

Between aspirate-dispense repetitions within a single `mix()` command, the pipette tip will move to the top of the well and then immediately move back down.

# Expected behavior

The pipette tip should stay at the same height for the entire `mix()` command.

# APIv2 only

This behavior is *not* present when uploading an APIv1 protocol. In APIv1, the pipette stays at the same height, as expected.

# Test case

```
metadata = {
    'apiLevel': '2.0'
}

def run(protocol):
    plate = protocol.load_labware('corning_96_wellplate_360ul_flat', '2')
    tiprack = protocol.load_labware('opentrons_96_tiprack_300ul', '1')
    pipette = protocol.load_instrument('p300_multi', 'right', tip_racks=[tiprack])
    pipette.pick_up_tip()
    pipette.mix(repetitions=4, volume=150, location=plate.wells_by_name()['A1'])
    pipette.return_tip()
```

# See also

- #780 maybe?

sfoster1, 2019-12-19:
This is called the "prep for aspirate" behavior. It's actually done in `aspirate` (which `mix`) calls, where if the current volume in the tip is 0, we move clear of the liquid and make sure the pipette plunger is at the empty position. This fixes a case where if you did this:

```
blow_out(A1)
aspirate(10, A1)
```

the plunger would move from the blowout position to the aspirate position while still in the liquid, accidentally overaspirating.

A more specific behavioral fix would probably be to append the move-to-empty behavior to the end of blowout.

btmorr, 2019-12-19:
I can see the argument for adding move-to-empty to the end of blowout. That said, it would remove a guarantee that `aspirate` currently provides (avoiding over-aspiration, as you pointed out). Could we provide the current guarantee instead by changing the conditional (https://github.com/Opentrons/opentrons/blob/161dd0bafbf011d52f2fcc48393762a4139765d7/api/src/opentrons/protocol_api/contexts.py#L822) from:

```
if self.current_volume == 0:
   # Make sure we're at the top of the labware and clear of any
   # liquid to prepare the pipette for aspiration
   if isinstance(dest.labware, Well):
       self.move_to(dest.labware.top())
```

to something like:

```
if self.current_volume == 0 and self.plunger_position < self.empty_position:
   # Make sure we're at the top of the labware and clear of any
   # liquid to prepare the pipette for aspiration
   if isinstance(dest.labware, Well):
       self.move_to(dest.labware.top())
```

The downside to this is that the hardware controller would have to expose more physical information to the protocol API (specifically, some notion of where the plunger is relative to some critical points--"is it below the empty position or not?"). On the other hand, this might be sensible info for the PAPI to be aware of, the same way that it can judge whether the next target well is in the same labware as the previous one in order to make movement decisions.

Thoughts, @sfoster1 @Laura-Danielle @ahiuchingau @mcous or anyone else?

SyntaxColoring, 2020-02-25:
A workaround is to break the `mix()` step into a bunch of separate `aspirate()` and `dispense()` steps, but there's a catch: you have to leave a little bit of liquid in the tip. Emptying the tip completely is what triggers the pipette "hop."

So, for example:

```python
pipette.aspirate(20)
pipette.dispense(19)
pipette.aspirate(19)
pipette.dispense(19)
```

PR #5128 (2020-03-04): fix(api): remove unnecessary pipette movement during mix
## overview
In APIv2, the pipette would move above the well between each aspirate-dispense repetitions during mix. The original logic for this movement is to reset the plunger position above liquid level after a blow-out command, to avoid over-aspiration in the subsequent `aspirate`. 

However, we currently evaluate the necessity of this action by checking the current volume of the pipette: if the volume is equal to 0, the pipette is always moved above the top of the well. Instead, we should check the plunger position and only move the p [... cut]
commit 88e18936 (2020-03-04): fix(api): remove unnecessary pipette movement during mix (#5128)

* fix(api): remove unnecessary pipette movement during mix

closes #4640
closed 2020-03-04 by ahiuchingau