Opentrons/opentrons #4649: Documented simulate method results in error

opatel0, 2019-12-19:
## overview
Our documented method to use the simulate function per API documentation results in an error.

## current behavior
The documented method for simulating protocols at the following link results in an error: [Simulating Your Scripts in Python](https://docs.opentrons.com/v2/writing.html#in-python)

Running `runlog = opentrons.simulate.simulate(protocol_file)` results in the following error: `TypeError: simulate() missing 1 required positional argument: 'file_name'`

Running `print(format_runlog(runlog))` results in the following error: `NameError: name 'format_runlog' is not defined`

## steps to reproduce
Save the following script as a .py on your computer:
```
from opentrons import labware, instruments

tiprack = labware.load('opentrons_96_tiprack_300ul','1')

pipette = instruments.P50_Single(mount='right',tip_racks=[tiprack])
plate = labware.load('corning_96_wellplate_360ul_flat', '2')

pipette.pick_up_tip()
for well in plate.wells():
	pipette.mix(3, 30, well)
```
Then open an interactive python shell and use the method outlined in the link above, written below for convenience
```
import opentrons.simulate
protocol_file = open('/path/to/protocol.py')
runlog = opentrons.simulate.simulate(protocol_file)
print(format_runlog(runlog))
```

## expected behavior
Running the assignment to `runlog` and the `print()` statement should not result in errors, and furthermore, the print statement should output something similar to when the same file is run through the command line with `opentrons_simulate file.py` 
Ex:
```
Picking up tip from well A1 in "1"
Mixing 3 times with a volume of 30.0 ul
	Aspirating 30.0 uL from well A1 in "2" at 1.0 speed
	Dispensing 30.0 uL into well A1 in "2" at 1.0 speed
	Aspirating 30.0 uL from well A1 in "2" at 1.0 speed
	Dispensing 30.0 uL into well A1 in "2" at 1.0 speed
	Aspirating 30.0 uL from well A1 in "2" at 1.0 speed
	Dispensing 30.0 uL into well A1 in "2" at 1.0 speed
Mixing 3 times with a volume of 30.0 ul
	Aspirating 30.0 uL from well B1 in "2" at 1.0 speed
	Dispensing 30.0 uL into well B1 in "2" at 1.0 speed
	Aspirating 30.0 uL from well B1 in "2" at 1.0 speed
	Dispensing 30.0 uL into well B1 in "2" at 1.0 speed
	Aspirating 30.0 uL from well B1 in "2" at 1.0 speed
	Dispensing 30.0 uL into well B1 in "2" at 1.0 speed
```

opatel0, 2019-12-19:
Current workaround to get the `runlog` statement running correctly is to instruct users to run the following sequence of commands as opposed to the documented sequence:
```
file_name = "path/to/file.py"
file = open(file_name)
runlog = opentrons.simulate.simulate(file, file_name)
```
Running `print(format_runlog(runlog))` results in the following error: `NameError: name 'format_runlog' is not defined`

Running `print(opentrons.simulate.format_runlog(runlog))` results in the following error: `TypeError: list indices must be integers or slices, not str`

Current workaround is to run a for loop:
```
for text in runlog[0]:
    print(text['payload']['text'])
```
However, workaround doesn't output with indented formatting

SyntaxColoring, 2020-01-21:
@Laura-Danielle Re-opening this because the merged fix seems incomplete.

See Ojas's comment above about the `TypeError` internal to `format_runlog()`.

PR #4660 (2019-12-20): fix(api): Fix simulate file from python shell
## overview

Closes #4649. 

## changelog

- Make `file_name` default to None in `simulate`
- Fixup the code in the docs and hopefully make it a little more clear to users which method they should use. (Open to suggestions here)

## review requests

Please review the documentation and see if it makes sense.
PR #4852 (2020-01-28): docs(api): fix docs around python interface to simulate
We need to make the tuple return of simulate clearer

Closes #4649
commit 62d73f05 (2020-01-28): docs(api): fix docs around python interface to simulate (#4852)

We need to make the tuple return of simulate clearer

Closes #4649
closed 2020-01-28 by sfoster1