Opentrons/opentrons #4650: App logs should be viewable from the More tab

SyntaxColoring, 2019-12-19:
# Background

As of 3.15.0 (?), we can reach app logs by going to **Help > View Opentrons App Logs.**

_I haven't tried this with any customers yet,_ but I suspect it might be a little too hidden.

All the other navigation instructions we have to give are within the app's Electron UI. Breaking out of that might take a bit of a cognitive shift for users, especially on macOS or Ubuntu, where the menu bar can be distant to the application window.

# Proposal

Put an app log button under **More > Advanced Settings** (or similar) to match the one in the menu bar.

# Additional premise questioning

We rarely need to get into the app directory, but when we do, it can be for reasons other than downloading logs. For example, we might need to enforce a totally fresh installation by gutting the entire Opentrons app folder.

The app log button doesn't serve this well, because it's not obvious on all platforms how to move up a directory. (I had to look it up on macOS.)

So, instead of an app log button, we might want an app folder button.

mattwelch, 2021-06-03:
Bug triage determination: LHL. Should be closed according to rubric, but keeping open as it's not actually a bug.

SyntaxColoring, 2021-06-10:
Closing my own issue here.

In my time on the Support team, I rarely had to direct customers to provide these particular logs. And when I did, I don’t remember them ever having trouble with it. This seems like a problem that only theoretically maybe needs solving.

closed 2021-06-10 by SyntaxColoring