Opentrons/opentrons #4666: Deadlock in opentrons_simulate

SyntaxColoring, 2019-12-21:
# Overview

`opentrons_simulate` consistently hangs for me at the very end of protocol simulation.

Interrupting with Ctrl+C gives a rather unhelpful traceback:

```
Exception ignored in: <module 'threading' from '/Users/maxpm/.pyenv/versions/3.8.0/lib/python3.8/threading.py'>
Traceback (most recent call last):
  File "/Users/maxpm/.pyenv/versions/3.8.0/lib/python3.8/threading.py", line 1388, in _shutdown
    lock.acquire()
KeyboardInterrupt: 
```

# Versions

## Python version

3.8.0 installed through pyenv.

## Output of `pip freeze`

```
aiohttp==3.4.4
aionotify==0.2.0
async-timeout==3.0.1
attrs==19.3.0
boto3==1.10.39
botocore==1.13.39
certifi==2019.11.28
chardet==3.0.4
docutils==0.15.2
idna==2.8
jmespath==0.9.4
jsonschema==3.2.0
multidict==4.7.1
numpy==1.17.4
opentrons==3.15.2
pipenv==2018.10.9
progress==1.5
pyrsistent==0.15.6
pyserial==3.4
python-dateutil==2.8.0
s3transfer==0.2.1
six==1.13.0
stm32loader==0.5.0
urllib3==1.25.7
urwid==1.3.1
virtualenv==16.7.8
virtualenv-clone==0.5.3
yarl==1.4.2
```

SyntaxColoring, 2019-12-21:
Slightly less unhelpful trace, from [hanging_threads](https://pypi.org/project/hanging-threads/):

```
Thread 123145564479488 "SynchAdapter thread for <<class 'opentrons.hardware_control.API'> using backend <class 'opentrons.hardware_control.simulator.Simulator'>>" hangs 
	File "/Users/maxpm/.pyenv/versions/3.8.0/lib/python3.8/threading.py", line 890, in _bootstrap
		self._bootstrap_inner()
	File "/Users/maxpm/.pyenv/versions/3.8.0/lib/python3.8/threading.py", line 932, in _bootstrap_inner
		self.run()
	File "/Users/maxpm/.pyenv/versions/3.8.0/lib/python3.8/threading.py", line 870, in run
		self._target(*self._args, **self._kwargs)
	File "/Users/maxpm/.local/share/virtualenvs/api-7zyHAEGB/lib/python3.8/site-packages/opentrons/hardware_control/adapters.py", line 76, in _event_loop_in_thread
		loop.run_forever()
	File "/Users/maxpm/.pyenv/versions/3.8.0/lib/python3.8/asyncio/base_events.py", line 563, in run_forever
		self._run_once()
	File "/Users/maxpm/.pyenv/versions/3.8.0/lib/python3.8/asyncio/base_events.py", line 1808, in _run_once
		event_list = self._selector.select(timeout)
	File "/Users/maxpm/.pyenv/versions/3.8.0/lib/python3.8/selectors.py", line 558, in select
		kev_list = self._selector.control(None, max_ev, timeout)
---------- Thread 4707003840 "MainThread" hangs   ----------
	File "/Users/maxpm/.pyenv/versions/3.8.0/lib/python3.8/threading.py", line 1388, in _shutdown
		lock.acquire()
```

SyntaxColoring, 2019-12-21:
`print()` logging everything that goes through `SynchronousAdapter.__getattribute__()`.

**On Python 3.6.9, where it terminates normally:**
```
__getattribute__('daemon')
__getattribute__('_initialized')
__getattribute__('_daemonic')
__getattribute__('is_alive')
__getattribute__('_initialized')
__getattribute__('_is_stopped')
__getattribute__('_started')
__getattribute__('_wait_for_tstate_lock')
__getattribute__('_tstate_lock')
__getattribute__('_is_stopped')
__getattribute__('join')
__getattribute__('_initialized')
__getattribute__('_started')
__getattribute__('_wait_for_tstate_lock')
__getattribute__('_tstate_lock')
__getattribute__('_stop')
__getattribute__('_tstate_lock')
[terminates normally]
```

**On Python 3.8.0, where it hangs:**

```
__getattribute__('ident')
__getattribute__('_initialized')
__getattribute__('_ident')
__getattribute__('name')
__getattribute__('_initialized')
__getattribute__('_name')
__getattribute__('ident')
__getattribute__('_initialized')
__getattribute__('_ident')
__getattribute__('name')
__getattribute__('_initialized')
__getattribute__('_name')
...
[repeats infinitely until I interrupt with Ctrl+C]
```

SyntaxColoring, 2019-12-21:
My current thoughts:

**First, the worker thread should probably be [daemonic](https://stackoverflow.com/questions/190010/daemon-threads-explanation).** Adding `daemon=True` to the constructor *does* resolve this problem for me! But, I'm not convinced that that's the only thing that's wrong.

**Second, the `__del__` method is suspicious.**
1. Garbage collection in Python [happens synchronously](https://www.python.org/dev/peps/pep-0556/#rationale), rather than in a separate thread.
2. Although stuff like [`thread_loop.call_soon_threadsafe()`](https://docs.python.org/3/library/asyncio-dev.html#concurrency-and-multithreading) is thread-safe (obviously), it's not necessarily reentrant.
3. Therefore, the GC can preempt the thread while it's in the middle of asyncio internals and holding a lock, call `__del__`, attempt to re-enter asyncio internals from `__del__`, and deadlock when it tries to re-acquire that same lock.

Basically, I don't think this kind of inter-thread cleanup is fundamentally possible from `__del__` without resorting to implementation-specific hacks. See [the warnings in the `__del__` documentation](https://docs.python.org/3/reference/datamodel.html#object.__del__).

sfoster1, 2020-01-02:
Interesting, thanks for looking into this.

@b-cooper is actually currently changing a lot about this subsystem. Most of its warts are from the fact that it's called from both synchronous and asynchronous contexts, and has an asynchronous interface that sometimes blocks when it talks to the smoothie. Instances can also be shared between different calling threads, which means you shouldn't really run the loops from there (or would require a lot of synchronization). The thread shouldn't be daemonic in general because of its use in contexts other than the one-and-done usage in the simulator and executor.

I think the short term fix here would be to make sure we're explicitly `join`ing the threads in both the simulator, the executor, and to a certain degree the main application (although it'll be a bit harder there for several extremely annoying reasons). The longer term is to change things so the ownership of a hardware controller thread is less confusing and we can do away with the issue; the longest term is to make the smoothie communication asynchronous so we don't actually need any threads anymore.

SyntaxColoring, 2020-01-02:
Explicit `join()`s will be tricky to get right in the face of exceptions.

Could we make it a `with`-compatible context manager, or would those "several extremely annoying reasons" preclude that?

sfoster1, 2020-01-02:
it would. the exception thing will be fine it just has to be handled at the callsite.

PR #4694 (2020-01-07): fix(api): properly clean up threads in opentrons_simulate, _execute
When these created synchronous adapter threads for hardware controllers, they
would never clean them up. This was fine on python 3.6 and 3.7 but not on
subsequent versions. We now clean them up properly.

Closes #4666 

## Testing
- simulate and (much less importantly since it always runs in a controlled environment) execute and make sure the program actually exits cleanly
closed 2020-01-07 by sfoster1