Opentrons/opentrons #4673: SCP CSV files to OT2 with Python scripts

ColtonGarelli, 2020-01-01:
## overview

I would like to add csv files to specific data folders on the OT2. I can connect to the robot with ssh commands (as explained in this article: https://support.opentrons.com/en/articles/3287453-connecting-to-your-ot-2-with-ssh ) and have created folders in the root/data folder. However, I am unable to connect programmatically (Python paramiko and scp packages, issuing bash command with os.subprocess) to connect and add files. This feels more like support rather than a bug or feature request, but maybe I'm doing everything correctly and something is going wrong.

## current behavior

Currently I can only create folders and files via ssh and scp commands through the terminal (`ssh -i ot2_ssh_key -o UserKnownHostsFile=/dev/null -o StrictHostKeyChecking=no root@Robot-IP` and `scp -i ot2_ssh_key -o UserKnownHostsFile=/dev/null -o StrictHostKeyChecking=no ./your_file.csv root@Robot-IP:/data/your_folder1`). 

Attempting to connect with `subprocess.call(bash_cmd.split())` with the same command listed above with and without sudo permission yields

Warning: Identity file ot2_ssh_key not accessible: No such file or directory.
Pseudo-terminal will not be allocated because stdin is not a terminal.
Warning: Permanently added 'Robot-IP' (ECDSA) to the list of known hosts.
ssh_askpass: exec(/usr/X11R6/bin/ssh-askpass): No such file or directory
Permission denied, please try again.
ssh_askpass: exec(/usr/X11R6/bin/ssh-askpass): No such file or directory
Permission denied, please try again.
ssh_askpass: exec(/usr/X11R6/bin/ssh-askpass): No such file or directory
root@Robot-IP: Permission denied (publickey,password).

Issuing this command with the full path to the key file yields the same response, but the no such file error is replaced with: 'Load key "file/path": invalid format'

Attempting to connecting with `paramiko.SSHClient().connect(hostname=Robot-IP, username='root', passphrase='passphrase')` throws an AuthenticationError with no arguments. I cannot make an ECDSA key (from the ot2_ssh_key file) using this package either. 

## steps to reproduce

I can access and add files via scp from the command line (though this article is out of date https://support.opentrons.com/en/articles/2519454-uploading-csvs-to-your-ot2-robot ). But doing the same with Python does not work. See above to reproduce.

I think this is more support than bug. I'd love to be able to run a script to add files to the robot from my local computer.

## expected behavior

I want to upload csv formatted plate maps that i can load into protocols that I am running on the robot (details: https://github.com/Opentrons/opentrons/issues/4637#issuecomment-567579451). Ideally, I'd be able to create a csv formatted plate map, load it to a folder on the OT2, and open that file in a protocol.

sfoster1, 2020-01-02:
Hey @ColtonGarelli , thanks for reaching out! Looks like you've got two specific questions here, about using `ssh` with `subprocess` and about using paramiko; and then of course the top-level question of "how do I put a file on my robot".

## how do I put a file on my robot?

Either use Paramiko (see below for how to do that) or use the robot's Jupyter notebook file upload process (the notebook is available on port 48888). If you upload files via the notebook, your code should look for them in `/var/lib/jupyter/notebooks`.

## `ssh` and `subprocess`

The issue here is that you're running `ssh` in a non-interactive context, and `ssh` is trying to do something interactive: get a password from you using its authentication agent. Given that you talk about a passphrase in the paramiko session, I think that's because (correct me if I'm wrong) you have a passphrase on your SSH private key.

 If you want to continue using a passphrase on your private key while invoking `ssh` through a `subprocess` call, you'll have to do one of the following:
- if you know exactly what to type (which you probably do; what ssh asks for is always the same), pass the password you need to enter as a string into the `stdin` argument of `subprocess.run` (see [python stdlib docs](https://docs.python.org/3.6/library/subprocess.html#subprocess.run))
- if you need to make decisions based on what `ssh` tells you, acquire the input pipe so you can talk to the ssh client using `subprocess.Popen` and `subprocess.Popen.communicate` ([see python stdlib docs](https://docs.python.org/3.6/library/subprocess.html#subprocess.Popen.communicate))
- for a minimal change, use something like [sshpass](https://linux.die.net/man/1/sshpass) instead of `ssh` directly

This is all a bit of a pain and I would recommend using paramiko for programmatic ssh access. You could also remove the passphrase from the key (or, really, generate a new one without the passphrase); but I won't go so far as to actively recommend that, presumably you have a good reason for wanting the passphrase.

## Paramiko

The error you're getting from Paramiko is because you haven't specified the private key file to use. You can do this with the `key_filename` argument:
`conn = paramiko.SSHClient().connect(hostname=Robot-IP, username='root', passphrase='passphrase', key_filename='/path/to/ot2_ssh_key')`. 
If the key is loaded into your system ssh agent if any, it _should_ automatically find it, but I haven't used that functionality in paramiko myself.

You'll also want to (before trying to connect) call `set_missing_host_key_policy(paramiko.client.WarningPolicy)` since the robot's host key currently changes every time it boots (not for long! We intend to fix this in 3.16). So the full sequence would be:

```python
import paramiko
client = paramiko.SSHClient()
client.set_missing_host_key_policy(paramiko.client.WarningPolicy)
connection = client.connect(hostname=robot_ip, username='root', passphrase='mypassphrase', key_filename='/path/to/ot2_ssh_key')
```

Hope this helps! Please let me know.

closed 2020-01-02 by sfoster1

ColtonGarelli, 2020-01-02:
Hi Seth. I tried the paramiko solution 
[image]

and got 

~/lib/python3.7/site-packages/paramiko/client.py:837: UserWarning: Unknown ecdsa-sha2-nistp256 host key for Robot-IP: b'52739d9f77d5d9b4a31081e3a79fe727'
  key.get_name(), hostname, hexlify(key.get_fingerprint())

('not a valid OPENSSH private key file',) <- args from the exception

I generated a new key, and used ssh to connect via terminal and still got this same error.

ColtonGarelli, 2020-01-03:
Okay I just got it to work. I think I had multiple keys on my computer and I think the second issue was related to that. I removed everything and regenerated keys and now it's working. Thanks again for your help! This should really improve our workflow.