Opentrons/opentrons #4675: API: BUG: In execute.execute(), if api_level not specified, do not assume APIv2

DerAndere1, 2020-01-02:
In APIv1 protocol files, `api_level` is usually not specified. For these protocols, `execute.execute()` wrongly assumes APIv2. I am not part of the Opentrons Inc team, so chances are that I am completely wrong. 

**Actual behaviour:**
If `api_level` is not specified in a protocol file, the current implementation of `execute.execute()` wrongly assumes APIv2.0.: 
https://github.com/Opentrons/opentrons/blob/5c3fae34c062680dae1f6391dc22ce7147fc4dd2/api/src/opentrons/execute.py#L279 

**Expected behaviour:**
In opentrons/execute.py, line 279 should be changed to:
```
    if getattr(protocol, 'api_level', APIVersion(1, 0)) < APIVersion(2, 0):
```
This way, APIv1 would be assumed by default and the APIv1-specific code would be executed.

sfoster1, 2020-01-02:
Hey @DerAndere1 ! Thanks for taking a look at this.

This naming is a bit misleading because the `protocol` object returned by `parse` is a named tuple with an `api_level` attribute that is scraped from the protocol when it is parsed. Whether the protocol is APIv1 or APIv2, the `api_level` attribute will always be present and (should) always be correct.

The `getattr()` call is actually there to handle running JSON protocols, which do _not_ have an associated `api_level` - they are kind of a separate mechanism.

DerAndere1, 2020-01-02:
I will close this as it is **NOT A BUG**. Can be reopened if you have a preferred way to handle obsolete issue reports.
Thanks.

closed 2020-01-02 by DerAndere1

sfoster1, 2020-01-02:
OK! Thanks for letting us know.