Opentrons/opentrons #4718: APIv2 protocols with errors are mislabeled as APIv1 on the upload page

SyntaxColoring, 2020-01-10:
The following APIv2 protocol, which has a typo, is incorrectly identified as APIv1 in the app.

```python
metadata = {'apiLevel': '2.0'}

def run(protocol):
    plate = protocol.load_labware('corning_384_wellplate_112ul_flat', 1)
    tip_rack = protocol.load_labware('opentrons_96_tiprack_10ul', 2)
    p10 = protocol.load_instrument('p10_single', 'left', tip_racks=[tip_rack])
    
    # Typo here.
    p10.transfe(5, plate['A1'], plate['A2'])
```

[image]

This can be confusing to users troubleshooting protocol problems.

When the typo is fixed, it is correctly identified as APIv2.

btmorr, 2020-01-13:
This is a side-effect of the RPC communication system with the robot, which is itself we are beginning to remove. Trying to fix this in RPC would assuredly cause other more critical failures.

@umbhau @mcous we should make sure HTTP APIv2 server is able to send back this kind of data correctly when a protocol is sent to the robot (maybe already on your radar, but wanted to ping so it doesn't get forgotten)

mcous, 2020-01-13:
As @btmorr mentioned, the app gets the API level from the protocol session. If the protocol fails simulation, the the API never hands the app a session, so the app defaults to 1.0

The most reasonable thing we can while we're still in RPC land is to blank out that field if the protocol simulation fails. It won't say anything, but it won't say anything wrong, either

mattwelch, 2021-06-03:
Bug triage determination: MLL. Closing as this will be deprecated with protocol engine work.

closed 2021-06-03 by mattwelch