Opentrons/opentrons #4719: API docs: sticky header obscures anchor-linked headings

SyntaxColoring, 2020-01-10:
# Overview

When following a link to a specific heading, that heading is buried behind the fat navbar.

This makes it difficult to show people function documentation.

For example, this is what it looks like 
when I try to directly link to [the Well.bottom() function reference](https://docs.opentrons.com/v2/new_protocol_api.html#opentrons.protocol_api.labware.Well.bottom):

[image]

# Expected behavior

The page should load with the linked-to heading positioned just below the navbar.

mattwelch, 2021-06-03:
Bug triage determination: MLH

mcous, 2023-02-07:
Moved to RSS-183. Fix PR up at #12115

closed 2023-02-07 by mcous