Opentrons/opentrons #4752: Protocol Designer: Build setup description is insufficient

henrynash, 2020-01-14:
## overview

The steps listed to build a development setup do not work

## current behavior

The current steps miss out the need to install webpack and the webpack-dev-server. In addition, the make dev command did not seem to launch the dev-server even after it was manually installed.

## steps to reproduce

Run the steps listed in the protocol-designer/README

## expected behavior

Either the main repo level makefile install should install web pack, or the protocol-designer makefile should do so (along with the webpack-dev-server)

mcous, 2020-01-17:
Hey @henrynash, `make install` does, in fact, install webpack et. al. if it completes successfully. Two questions:

- Did you follow the [devemopment setup guide in CONTRIBUTING.md](https://github.com/Opentrons/opentrons/blob/edge/CONTRIBUTING.md#development-setup)
- Do you happen to be using macOS 10.14 or later?
    - `make` in macOS 10.14 appears (to us) to have some serious bugs related to finding binaries in `$PATH`, so `make dev` (and others) somehow fails to find executable files in `node_modules/.bin` even though their in the PATH
    - Long story short, all of our macOS devs have switched to using GNU make, installable via `brew` because after much hair pulling none of us could get it to work

Regardless, it does look like PD's README.d needs a callout to CONTRIBUTING.md, and CONTRIBUTING.md probably needs to recommend GNU make specifically

henrynash, 2020-01-17:
Ah, right - yes, I think that maybe the cause (Mac 10.15).  Agree some additional doc changes would help - I'd be happy to submit a PR for that (as I am wanting to start contributing and happy to chop wood, carry water to start)

mcous, 2020-01-17:
Any PRs (docs especially) would be much appreciated!

henrynash, 2020-01-19:
As an aside, I think I found a good fix for the Makefile issues on macOS - see #4809

mcous, 2020-11-19:
Closing as resolved by makefile fixes listed above

closed 2020-11-19 by mcous