Opentrons/opentrons #4756: PD: Add Air Gap functionality to PD

mikec116, 2020-01-14:
## User Story
As a PD user, I want to be able to add an air gap after aspiration to have better results with my transfers. 

## Acceptance Criteria
TBD

## UI Design
[Add Zeplin link]

IanLondon, 2020-01-22:
Marking blocked b/c we have to solve the liquid tracking problems this introduces before getting into it. I think @howisthisnamenottakenyet documented them a year ago but we'd have to dig it up. Random notes from what I remember:

- We might avoid liquid tracking problems by requiring blowout if you use air gap.
- We need errors for over-aspirating when aspirated volume plus the air gap exceed pipette/tip capacity (kind of like disposal volume in many-to-one path aka distribute).
- I think air gap is allowed for 1:1 transfer, not allowed for one-to-many (distribute), and potentially could be allowed for many-to-one (consolidate)?

IanLondon, 2020-06-30:
Closing b/c this is stale, we'll soon make a new ticket as part of air gap + delay project

closed 2020-06-30 by IanLondon