Opentrons/opentrons #4757: Wired MAC addresses should be predictable before connecting

SyntaxColoring, 2020-01-15:
# Current behavior

An OT-2's wired MAC address is only visible once you're already connected to the OT-2. This makes it somewhat difficult to set up wired DHCP networks that assign a static IP addresses based on MAC. Such networks are one way to work around failing robot auto-discovery.

# Desired behavior

Some way to predict what the OT-2's wired MAC address will be, that doesn't rely on establishing any kind of connection to it.

Maybe derive it from the serial number.

# Migration

People have networks set up relying on their OT-2s' current MAC addresses. We need a transition plan so that those people aren't forced to reconfigure their networks when this gets fixed.

mcous, 2020-01-15:
Technically, I don't think we can just pick MAC addresses out of thin air. IIRC, if we were to generate these things in software and derive them, we'd need to purchase an [OUI](https://en.wikipedia.org/wiki/Organizationally_unique_identifier) from the IETF at minimum

The MAC address on board the Pi itself should be stable. I personally think this is the kind of thing that is best addressed by a process change to:

- Grab the wired and wireless MAC addresses at device provisioning time
    - Maybe from the Rasp. Pi's box?
    - Acknowledging that we'd need to fix #4758
- Print those MAC addresses on the OT-2 sticker like any other internet device

SyntaxColoring, 2020-01-15:
A sticker sounds totally acceptable to me. But there might be product design considerations to compromise against.

We would also need to revise Support's SD card replacement process.

SyntaxColoring, 2021-03-26:
This is not addressed in the OT-2 refresh (at the time of writing, anyway).

I'm iceboxing this because we don't know how much it actually matters.

The release of the OT-2 refresh, with its real Ethernet port, might affect customer demand for this feature, but we don't know by how much. When we have more field experience with the OT-2 refresh, we can revisit this and either put it in the backlog or confidently close it as not worth implementing.

mattwelch, 2021-06-03:
Bug triage determination: MHM

mcous, 2021-12-10:
Gonna make an executive decision and close this one. Adding the wired MAC address to the label was floated internally but got no uptake. With the OT-2R's exposed Ethernet port and the general drop in connectivity support cases (in terms of what gets surfaced somewhere I can see it), I don't think this ticket will go anywhere

closed 2021-12-10 by mcous