Opentrons/opentrons #4758: Wireless MAC addresses should be even more stable

SyntaxColoring, 2020-01-15:
# Current behavior

* The robot uses a different MAC address for each wireless network.
* Those MACs aren't predictable. You have to actually connect to a network for the app to tell you what MAC address the robot uses for it.

This makes it somewhat difficult to set up networks that have whitelisting or static IP assignment based on MAC.

# Desired behavior

* There is One Wireless MAC Address To Rule Them All. It doesn't change from network to network.
* That Wi-Fi MAC address is visible somewhere *before connecting to Wi-Fi.* It's acceptable if it requires a wired connection to see, because you need a wired connection to set up Wi-Fi anyway.

# Migration

People have networks set up based on their OT-2s' current MAC addresses. They should be grandfathered in; we shouldn't make them reconfigure their networks when this is fixed.

# Prior issues

* #3175 
* #4121

mcous, 2020-04-28:
Related ticket: #5200

mcous, 2020-05-27:
Wi-Fi MAC address appears in code and in actual behavior to be stable? Investigating

SyntaxColoring, 2020-05-27:
> Wi-Fi MAC address appears in code and in actual behavior to be stable? Investigating

Hm!

If it turns out that it *is* stable, let's be careful about closing this issue. One of the expected behaviors is:
> [The] Wi-Fi MAC address is visible somewhere before connecting to Wi-Fi...

If we close this ticket, we should open a new ticket for that. (In hindsight, that should have been in a separate ticket, to begin with.)

mcous, 2020-06-25:
Updated the original comment. We do not need a separate ticket for displaying MAC in the app, because the MAC has always been displayed in the app.

When connected to Wi-Fi (no matter the network), the MAC address is stable. When disconnected from Wi-Fi, the robot reports a random MAC address for the Wi-Fi interface to the app

SyntaxColoring, 2020-06-25:
In addition to needing to connect to a wireless network to see the correct wireless MAC address:

It might be a good idea to close and reopen the Opentrons App before checking the MAC.

I'm not 100% sure this actually matters, but I *think* I've seen the Opentrons App show stale IP addresses. (There's no issue ticketed for this, yet.) The same thing might happen for MAC addresses.

mcous, 2020-06-25:
> I'm not 100% sure this actually matters, but I think I've seen the Opentrons App show stale IP addresses. (There's no issue ticketed for this, yet.) So it might do the same thing for MAC addresses.

The app polls the `/networking/status` endpoint on something like a 5 or 10 second interval. The IP and MAC come from that endpoint. It'll take a second after reconfiguring Wi-Fi for that info to update

closed 2020-07-06 by sanni-t