Opentrons/opentrons #4761: The error for a wrong Wi-Fi password should be less cryptic

SyntaxColoring, 2020-01-15:
# Current behavior

1. Try connecting to a Wi-Fi network with a wrong password.
2. The app will give you a message saying "Secrets were required but not provided." This cute but cryptic message apparently bubbles up from NetworkManager.

# Desired behavior

For the common cases of:

* Getting the password wrong for a WPA2 Personal network
* Getting any of the credentials wrong for a WPA2 Enterprise network

We should have copy to the tune of:

> There was a problem connecting to that network. Check your credentials and try again.
>
> *Hint: Try the **Show password** checkbox!*

mattwelch, 2021-06-03:
Bug triage determination: MLL

SyntaxColoring, 2024-02-27:
Closing as low-priority and sort-of-fixed.

This is how the message shows up today:

[image]

There's a lot of unhelpful technobabble at the top, but it *does* now say to check your network credentials, in the middle.

closed 2024-02-27 by SyntaxColoring