Opentrons/opentrons #4763: [Discussion] Simplify API for manipulating and moving to deck coordinates

SyntaxColoring, 2020-01-15:
# Background

A few customers have been unable to figure out how to move to deck coordinates. This is partially because of incomplete documentation, but also, the interface is surprisingly layered.

This is how it's done:

```python
from opentrons import types
pipette.move_to(location=types.Location(point=types.Point(x, y, z), labware=None))
```

(Args provided by name for the sake of discussion.)

There's a lot of boilerplate, here.

# Proposed changes, off the top of my head

* I don't want to go through internal types like `Point` or `Location` unless I want to use their specific features (and those features couldn't have been elegantly implemented at the function level).
* It should be difficult to use the API incorrectly. If I need to "really know what I'm doing" to provide a sane value for an argument, don't make me provide a value for that argument.

A good simplified interface could look like:

```python
pipette.move_to((x, y, z))
# Or maybe:
pipette.move_to(Point(x, y, z))
# Or maybe:
pipette.move_to(Location(x, y, z))
```

But probably not:

```python
pipette.move_to(Location(Point(x, y, z)))
```

# Discussion

We should agree on exactly what compromises to make. I appreciate that a goal of APIv2 is to keep function signatures as simple as possible.

The semantics of `Location` might make this a little nuanced.

`Location` has `loc.move(point)` (which should maybe also be spelled `loc + point` or at least `loc.plus(point)`, but that's another story), effectively making `Location` the closest thing that the API has to a general-purpose displacement vector type. Should we keep going in that direction?

mcous, 2020-01-15:
If this were implemented in PD, we would create a new atomic command for it. Accordingly, if the ask is "as a user, I'd like to move the pipette to an arbitrary location in deck coordinates", then I think the best thing to do would be to have a `pipette.move_to_location` method rather try to overload the existing `move_to` with new or changed behavior

SyntaxColoring, 2020-01-16:
@mcous That's a good way of looking at it. 👍

theosanderson, 2020-09-30:
So I hadn't seen this and opened #6655 which proposed `pipette.move_to(Point(x, y, z))` as above. I definitely see @mcous 's point on the possibility of a separate atomic command.

On the other hand, I actually think there is an argument for at some stage adding this functionality to as many of the pipette functions as possible (where it doesn't create a conflict with something else). For example there are several reasons one might want to eject tips at a particular point (creating a command to randomly position them in the trash, or drop them into a custom chute at the front of the robot). Given that I would kind of argue for `move_to` to be able to take either type. But this is definitely not a strong preference. If you folks decide on how you'd like to move forward I'll have this on the todo list.

Thanks a lot for the PR review @sfoster1

closed 2022-09-30 by bndo