Opentrons/opentrons #4772: Motor controller errors should be more self-descriptive

SyntaxColoring, 2020-01-16:
# Background

Motor controller errors (a.k.a. "Smoothie errors") are a bit of a black box to Support – at least, to me – because they're not self-descriptive. I have to do Slack archeology most times I see one.

# Examples

* WTF is a "hard limit"? (It's the limit switch being hit when not homing.)
  * Instead, something like: `Error: -Z limit switch unexpectedly triggered`
* WTF is a "homing fail"? (It's the limit switch *not* being hit when homing.)
  * Instead, something like: `Error: Expected to trigger -Z limit switch, but it wasn't within 5s`
* WTF is "alarm lock"? (I still don't know.)
* WTF is this G-code doing? (Some of it is standard, but not all!)
  * G-code seems like the wrong level of abstraction for most of our work. I take for granted that the G-code layer is working properly. I care about the higher-level command that failed, like "with current 1.0, moved -Z by amount 10."

# Desired behavior

At a high level, messages should:
* Be meaningful to a person who's familiar with the platform's fundamentals but not all of its internals.
* Be unambiguous.
* Be appropriately specific.

SyntaxColoring, 2021-02-10:
Since this issue was written, we've solved this for our own Support team not by changing the error messages, but instead by writing up a cheat sheet to help interpret the existing messages. This seems to be good enough.

That cheat sheet is currently internal-only because:

* Making it good enough to be public would take nonzero work and thought.
* I kind of doubt that any user cares about errors *at the level that this issue talks about.* Like, even if we successfully taught a user to read the error message, understand what the phrase "hard limit" means, and mentally parse the line of G-code to understand what the robot was doing when the error happened...how does that help them? They'll still need Support's help to resolve the problem of broken hardware or whatever.
  * I mean, maybe some advanced users would find it interesting. But I think those same advanced users could search our codebase for `SmoothieError` and figure it out on their own? 

But, if this seems wrong, I'm happy to reconsider.

-----

> Messages should call it the "stepper motor driver" or "motor controller," not "Smoothie."

This still irks us, but I think it belongs in its own issue.

closed 2021-02-10 by SyntaxColoring

Akuli, 2021-05-24:
Is "if you get G-code errors then contact support" documented somewhere? I spent some time googling, and I realized I should contact the support when I found this issue.

SyntaxColoring, 2021-05-24:
@Akuli I think I assumed that if someone gets a G-code error, it's ultimately surfaced with a scary message that encourages people to reach out to Support.

What particular G-code error did you get, and where? Maybe we can improve how that error is presented.

Akuli, 2021-05-24:
When I ask the robot to home or move somewhere, it makes a weird brrr noise instead of actually moving. I found the G-code error from the logs. I first assumed that it's a software problem, but I'm no longer sure about it. I contacted support.

<details><summary>Part of api.log</summary>

```
May 24 11:20:04 misty-river opentrons-api[239]: rpc built call move_to on <opentrons.api.calibration.CalibrationManager object at 0x718d1070>
May 24 11:20:04 misty-river opentrons-api[239]: this is the first calibration action, homing
May 24 11:20:04 misty-river opentrons-api[239]: Homing axes XYZAXZA in sequence ['AZ', 'X', 'Y']
May 24 11:20:10 misty-river opentrons-api[239]: Homing axes B in sequence ['B']
May 24 11:20:15 misty-river opentrons-api[239]: Homing axes C in sequence ['C']
May 24 11:20:15 misty-river opentrons-api[239]: alarm/error outside hard halt: ALARM: Hard limit -C
May 24 11:20:16 misty-river opentrons-api[239]: alarm/error: command=G0 C-1

, resp= ALARM: Hard limit -C
May 24 11:20:18 misty-river opentrons-api[239]: alarm/error outside hard halt: ALARM: Hard limit -C
May 24 11:20:18 misty-river opentrons-api[239]: alarm/error: command=M907 A0.1 B0.3 C1.0 X0.3 Y0.3 Z0.1 G4 P0.005 G0 C-14.5

, resp= ALARM: Hard limit -C
May 24 11:20:18 misty-river opentrons-api[239]: Homing after alarm/error
May 24 11:20:18 misty-river opentrons-api[239]: Homing axes C in sequence ['C']
May 24 11:20:20 misty-river opentrons-api[239]: Move failed
Traceback (most recent call last):
File "usr/lib/python3.7/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 1072, in _send_command
File "usr/lib/python3.7/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 1108, in _send_command_unsynchronized
File "usr/lib/python3.7/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 1147, in _handle_return
opentrons.drivers.smoothie_drivers.driver_3_0.SmoothieError: SmoothieError: None returned ALARM: Hard limit -C

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "usr/lib/python3.7/site-packages/opentrons/hardware_control/api.py", line 1093, in _move
File "usr/lib/python3.7/site-packages/opentrons/hardware_control/controller.py", line 118, in move
File "usr/lib/python3.7/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 1642, in move
File "usr/lib/python3.7/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 1089, in _send_command
opentrons.drivers.smoothie_drivers.driver_3_0.SmoothieError: SmoothieError: M907 A0.1 B0.3 C1.0 X0.3 Y0.3 Z0.1 G4 P0.005 G0 C-14.5

returned ALARM: Hard limit -C
May 24 11:20:20 misty-river opentrons-api[239]: Exception during RPC call:
Traceback (most recent call last):
File "usr/lib/python3.7/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 1072, in _send_command
File "usr/lib/python3.7/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 1108, in _send_command_unsynchronized
File "usr/lib/python3.7/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 1147, in _handle_return
opentrons.drivers.smoothie_drivers.driver_3_0.SmoothieError: SmoothieError: None returned ALARM: Hard limit -C

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "usr/lib/python3.7/site-packages/robot_server/service/legacy/rpc/rpc.py", line 269, in make_call
File "usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
File "usr/lib/python3.7/site-packages/robot_server/service/legacy/rpc/rpc.py", line 259, in call_and_serialize
File "usr/lib/python3.7/site-packages/opentrons/api/util.py", line 32, in decorated
File "usr/lib/python3.7/site-packages/opentrons/api/calibration.py", line 42, in decorated
File "usr/lib/python3.7/site-packages/opentrons/hardware_control/adapters.py", line 56, in call_coroutine_sync
File "usr/lib/python3.7/concurrent/futures/_base.py", line 432, in result
File "usr/lib/python3.7/concurrent/futures/_base.py", line 384, in __get_result
File "usr/lib/python3.7/site-packages/opentrons/hardware_control/api.py", line 681, in home
File "usr/lib/python3.7/site-packages/opentrons/hardware_control/api.py", line 647, in _do_plunger_home
File "usr/lib/python3.7/site-packages/opentrons/hardware_control/api.py", line 993, in _move_plunger
File "usr/lib/python3.7/site-packages/opentrons/hardware_control/api.py", line 1093, in _move
File "usr/lib/python3.7/site-packages/opentrons/hardware_control/controller.py", line 118, in move
File "usr/lib/python3.7/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 1642, in move
File "usr/lib/python3.7/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 1089, in _send_command
opentrons.drivers.smoothie_drivers.driver_3_0.SmoothieError: SmoothieError: M907 A0.1 B0.3 C1.0 X0.3 Y0.3 Z0.1 G4 P0.005 G0 C-14.5

returned ALARM: Hard limit -C
```

</details>

Also, given an ssh connection to the robot, it would be nice to `tail -f` the log file. I found a few files named `api.log` on the robot, but none of them seemed to contain the full content of what the "Download logs" button gives. That way I could see the errors as they happen instead of looking at timestamps. Should I make a separate issue about this or is it just a matter of knowing where the correct `api.log` is?

SyntaxColoring, 2021-05-24:
I see. So the Opentrons App isn't showing you an error. You're doing your own debugging, seeing the phrase "hard limit" in the internal logs, and wondering what that means.

In that case, to answer your question: no, we haven't documented anywhere that you should contact Support if you see a hard limit error in the internal logs.

SyntaxColoring, 2021-05-24:
> Also, given an ssh connection to the robot, it would be nice to tail -f the log file. I found a few files named api.log on the robot, but none of them seemed to contain the full content of what the [...]

Yeah, I'm happy to help with that in a separate issue. Feel free to mention @SyntaxColoring in it to ping me.

Akuli, 2021-05-31:
This is the first error I got, and it's from the app:

[image]

As a software developer, when I see an error like this, my first instinct is to get more information about it (from logs, in this case), and that's especially true when the support opens 4PM in my timezone and has about 1h of waiting. But the support resolved my problem and the error does say "contact support", so there's nothing to be done here.