Opentrons/opentrons #4796: Pipette calibration is always skipped when running an API v1 protocol on v. 3.15.2

Cin316, 2020-01-17:
## overview

Pipette calibration is always skipped when running an API v1 protocol on v. 3.15.2.  API v2 protocols run normally, with pipette calibration occurring for them.  This behavior was not observed in prior versions (v 3.10.3).

## current behavior

Every time an API v1 protocol is run on an OT-2 running server version 3.15.2 and using app version 3.15.2, pipette calibration is completely skipped.  There is no prompt to remove the trash pin cover and calibrate the pipette heads.  It is skipped entirely, and labware calibration continues as normal.  Protocols do work normally, but I feel that missing this calibration data may cause drift over time.

This was observed with a protocol that uses a P300_multi and a P300_single Gen1 pipettes.

## steps to reproduce

Create a protocol that uses both pipettes.  Attach Gen1 P300_multi and P300_single pipettes to the OT-2.  Attempt to run a API v1 protocol.  Observe that pipette calibration is skipped.

## expected behavior

The app should prompt you to remove the trash bin cover and place a pipette tip on the pipette. 
 Then, pipette calibration should occur.

mcous, 2020-01-17:
@Cin316 I just tested this on one of our 3.15.2 robots using an APIv1 protocol (`api/tests/opentrons/data/dinosaur.py` from the repository, specifically) and was not able to reproduce this issue

Would you mind sending along your protocol (or a similar protocol that reproduces your issue)?

Cin316, 2020-01-17:
Unfortunately, our protocol contains proprietary information and I cannot release it.  Additionally, today is my last day working in this lab, so I don't have time to try and make a minimal, reproducible example protocol.  Hopefully someone else from this lab can give you more information in the future if it continues happening.  We've decided to skip trying to update and remain on version 3.10.3. 

Just keep this issue in mind in case if you get more reports of this problem.  It looked like the robot thought there were no pipettes used for the protocol, so it skipped calibrating them completely.

mcous, 2020-01-17:
Oh, I see. As of 3.15 and Protocol API v2, all pipettes spec'd in the protocol are considered "used" regardless of what the inconsistent simulation use-checking logic. I'm going to close this ticket accordingly.

Without upgrading past 3.10, you will not see an improvement in this behavior. As a workaround, you could try adding a "dummy" pickup and return tip for each pipette to force it to appear in the calibration list. But honestly, especially on 3.10, you may be better off skipping tip probe, as it has a history in previous releases of adding error to the tolerance stack rather than removing it

Cin316, 2020-01-17:
@mcous Both pipettes were used in that protocol, however the software incorrectly identified them as "not used".  There were no checks next to their items on the calibration list.

mcous, 2020-01-17:
@Cin316 yup, I understand what you're saying! All versions of robot software prior to 3.15 run the risk of incorrectly identifying pipettes as "not used" depending on weird stuff about how the protocol is written. This bug was fixed as part of the 3.15 release and Protocol API v2.

Without either upgrading the robot's software or adding workarounds to the protocol to force the buggy software to recognize the pipette as used, this incorrect behavior will likely persist for your robot + protocol

Cin316, 2020-01-17:
I'm saying that that bug was introduced with 3.15.2 for our case.  In API v1 and 3.10.3, it correct identifies which pipettes we use and calibrates them.  In API v2 and 3.15.2, it correctly identifies which pipettes we use and calibrates them.  In API v1 and 3.15.2, though, it incorrectly fails to identify that we use pipettes, and skips pipette calibration entirely.

mcous, 2020-01-17:
My apologies, thank you for your clarification bearing with my misunderstanding of the issue.

I'm sorry that you're not able to provide a reproduction protocol, because that would be helpful in this case seeing as the PAPIv1 protocols we have handy do not exhibit this behavior. I'll add this to the backlog. Without a reproduction protocol, I'm not sure we'll necessary have the resources to devote to fixing this, but like you said, maybe we'll get more reports

mcous, 2021-12-10:
No longer applicable due to the deprecation and removal of PAPIv1

closed 2021-12-10 by mcous