Opentrons/opentrons #4807: Make copyright status explicit in the opentrons repository / source code

DerAndere1, 2020-01-18:
## overview

It would make people more comfortable if you could make the copyright and license status of the opentrons source code explicit. Currently there is an Apache 2.0 license file at the root of the repository. However that license is not applied properly. 
 
## current behavior

Copyright notice and license info boilerplate missing.

## steps to reproduce

use the Github search inside this repository to find instances of "Copyright". 
->No copyright statement is found

## expected behavior

Copyright statement and license boilerplate, at least somewhere in this repository, in one or more prominent places. If you want to keep all copyright and license information in one place, add a preface text at the beginning of the license file that includes a notion to what parts of the repository the Apache 2.0 License applies, followed by the required boilerplate and copyright notice, followed by the Apache 2.0 License text itself. Or add the boilerplate in the beginning of some or most files.
Here is the relevant excerpt from the Apache 2.0 license appendix:

>    To apply the Apache License to your work, attach the following
      boilerplate notice, with the fields enclosed by brackets "[]"
      replaced with your own identifying information. (Don't include
      the brackets!)  The text should be enclosed in the appropriate
      comment syntax for the file format. We also recommend that a
      file or class name and description of purpose be included on the
      same "printed page" as the copyright notice for easier
      identification within third-party archives.

   Copyright [yyyy] [name of copyright owner]

   Licensed under the Apache License, Version 2.0 (the "License");
   you may not use this file except in compliance with the License.
   You may obtain a copy of the License at

       http://www.apache.org/licenses/LICENSE-2.0

   Unless required by applicable law or agreed to in writing, software
   distributed under the License is distributed on an "AS IS" BASIS,
   WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
   See the License for the specific language governing permissions and
   limitations under the License.

mcous, 2020-01-21:
Thanks for the thoughts! We'll get that boilerplate added to the LICENSE file. Aside from that missing statement, though, I think we're pretty explicit in how everything is Apache-2.0 licensed:

- All JS projects include `"license": "Apache-2.0"` in their `package.json` manifest files
- All Python projects include `license="Apache 2.0"` in their `setup.py` files
    - They also have `package.json` files with `"license": "Apache-2.0"` to behave in our monorepo
- The GitHub repository highlights that the project is Apache-2.0 in the header

[image]

So, if you're worried, please rest assured that everything in here is Apache 2.0 licensed!

DerAndere1, 2020-01-22:
Thanks. commit d4250b6 solves this issue. 
I appreciate that you have always been very clear about the **intended** license. It's just that the License required this explicit notice.

mcous, 2020-01-24:
@DerAndere1 I'm actually unable to find any documentation that states this sort of notice is required for the license to be valid. I'm of the personal opinion that it is optional, given:

- Copyright is granted automatically to any work, regardless of whether the creator states that fact
- We have multiple "notices" (via `package.json`s etc) that attach the included Apache-2.0 license to this work

Are you able to point me to any documentation that states that a notice is strictly required?

DerAndere1, 2020-01-27:
The appendix of your LICENSE file mentions no alternive method of licensing source code. But now I see that https://www.apache.org/licenses/LICENSE-2.0 has an alternatve apendix. Still, that one line of plain-text copyright notice in the NOTICE file (or alternatively a copyright notice in each file) ensures attribution and documentation of copyright in derivative code, even if only a single file was copied and modified.

PR #4859 (2020-01-27): docs(repo): add NOTICE file with copyright and warranty statements
## overview

This PR makes the copyright status of this repository slightly more explicit. 

Closes #4807. See issue for details and discussion

## changelog

- docs(repo): add NOTICE file with copyright and warranty statements

## review requests

- [ ] Looks OK + legal
commit cfd103fc (2020-01-27): docs(repo): add NOTICE file with copyright and warranty statements (#4859)

Closes #4807
closed 2020-01-27 by mcous