Opentrons/opentrons #4808: Makefiles do not work properly on macOS

henrynash, 2020-01-18:
## overview

Many of the Makefiles have problems finding execuatbles on macOS 

## current behavior

If you use the standard macOS make command, often the make targets will fail due to not being able to find executables that are, in fact, in the PATH. This is due to PATH not being properly honored in shell invocations in macOS make.

## steps to reproduce

Run any of the project Makefiles (e.g. protocol-designer)

## expected behavior

Even though we recommend GNU make, it would be better if the standard make worked on macOS. This can be achieved by explicitly setting the PATH env variable in the SHELL command in the Makefile, e.g.:

```
PATH := $(shell cd .. && yarn bin):$(PATH)
SHELL := env PATH=$(PATH) /bin/bash
```

henrynash, 2020-01-18:
I'll submit a PR for fixes to the Makefiles.

SyntaxColoring, 2020-01-21:
I don't think I've seen this problem.

`make --version` reports "GNU Make 3.81" for me, so it looks like the standard macOS `make` command *is* GNU Make. Do you see something different?

henrynash, 2020-01-21:
So actually you are right that make does have the GNU make...but there is something screwy with how make on macOS invokes shell commands (that ignores PATH modifications within the makefile). Here's a simple example:

(setup)

```
mkdir test
echo "echo hello" > test/mytest
chmod a+x test/mytest
```

(now create a Makefile containing)

```
PATH := ./test:$(PATH)
SHELL := /bin/bash

.PHONY: all
all:
	echo $(PATH)
	mytest

```
if you run make, it will fail with:

`make: mytest: No such file or directory`

Note that the PATH echo'd by the Makefile DOES have ./test in it!

However if you change the SHELL line in the Makefile to:

`SHELL := env PATH=$(PATH) /bin/bash`

...it works.

henrynash, 2020-01-21:
So in the Opentrons environment, unless the dependencies at the repo level are installed globally (or indeed the PATH is set permanently to include the `yarn bin` directory), then the project makefiles will fail since they update the PATH to include the `yarn bin` directory for the purpose of executing items installed at the repo level.

mcous, 2020-01-22:
@SyntaxColoring everyone in the office running macOS 10.14 or later has run into this (myself included)

@henrynash this is so weird, right? The craziest thing that I can't figure out is this:

```shell
#!/usr/bin/env bash
# ./node_modules/.bin/shell-script
echo "shell script success!"
```

```makefile
# ./Makefile
SHELL := /bin/bash
PATH := ./node_modules/.bin:$(PATH)

.PHONY: broken
broken:
	env
	shell-script

.PHONY: works
works:
	env && shell-script

.PHONY: also_works
also_works:
	sleep 0 && shell-script
```

```shell
make broken
#> env
#> ...
#> PATH=./node_modules/.bin:...
#> ...
#> shell-script
#> make: shell-script: No such file or directory
#> make: *** [all] Error 1
```

```shell
make works
#> env && shell-script
#> ...
#> PATH=./node_modules/.bin:...
#> ...
#> shell script success!
```

```shell
make also_works
#> sleep 0 && shell-script
#> shell script success!
```

SyntaxColoring, 2020-01-22:
@mcous My suspicion is that for simple commands, `make` is optimizing away the subshell invocation and executing the command directly, triggering the bug. (This is [supported by this stranger on Stack Overflow](https://stackoverflow.com/a/21709821/497934) so it must be true!)

See this note [in their docs](https://www.gnu.org/software/make/manual/html_node/Execution.html), emphasis mine.

> When it is time to execute recipes to update a target, they are executed by invoking a new sub-shell for each line of the recipe...**(In practice, make may take shortcuts that do not affect the results.)**

I guess they missed a case in the "do not affect the results" criteria.

-----

Also. I'm sure this has occurred to you guys, but in case it hasn't: instead of globally modifying `PATH`, could we just always execute Yarn-installed commands with `yarn run foo` or however they spell it? That seems a bit more explicit and robust.

henrynash, 2020-01-22:
@SyntaxColoring Agreed...it is pretty weird! (and it is still a problem on 10.15). In fact you can make the line as simple as:

```
works:
	: && shell-script
```

However, my suggestion is that it would be best to have something standard in the pre-amble in each Makefile to fix this...rather than rely on future Makefile modifiers knowing that this problem exists on macOS. This is why I proposed the `SHELL := env PATH=$(PATH) /bin/bash` solution - since it means authors can then ignore the issue for the main contents of the Makefile.

PR #4809 (2020-02-05): build(make): ensure SHELL includes PATH in its environment
## overview

make on macOS does not seem to honor updates to PATH within the Makefile for shell invocation. This can be resolved be setting the PATH env variable as part of the SHELL command. This change is benign for GNU make.

fix #4808

## changelog

All Makefiles have been updated to modify the existing line that set the SHELL env variable to include the PATH.

## review requests

I have tested this using standard make on macOS (10.15) and GNU make on Ubuntu linux. Some tests on other configurations would provide confidence that th [... cut]
commit 6423f719 (2020-02-05): build(make): ensure SHELL includes PATH in its environment (#4809)

Fixes #4808
closed 2020-02-05 by mcous