Opentrons/opentrons #4810: Labware calibration: "Continue moving to tip rack" button sometimes does nothing

SyntaxColoring, 2020-01-21:
# Current behavior

The "Continue moving to tip rack" button in labware calibration sometimes silently fails.

1. The user clicks the "Continue moving to tip rack" button during labware calibration.
2. The loading spinner briefly appears over the tip on the deck layout view, then disappears.
3. Nothing happens. The robot doesn't move. The user can't continue with labware calibration.

# Expected behavior

Either the robot should move to the tip rack, or we should see an error message explaining why the robot isn't moving to the tip rack.

# Possible causes

This *may* have something to do with a bad deck calibration. Redoing deck calibration has resolved the problem for one customer.

See this issue's Slack thread for case references.

mcous, 2020-01-21:
The app collects the error from the API and does not display it. We can display the error, but as far as the error message goes, while calibration processes are on the legacy RPC API they will be mostly, if not completely, non-user-actionable.

The most common error to cause this behavior will be fixed by #4802

mcous, 2020-02-13:
Oh also, unrelated to this fix for this ticket, but in the logs referenced above, that specific bug was fixed in #4614 and released in 3.15.2

SyntaxColoring, 2020-05-03:
Okay, so the issue here is that internal problems – whatever they might be – are silently swallowed instead of being displayed in an error dialog or whatever.

Rather than rewriting this ticket, I've created a new one: #5551. Closing this in favor of that.

closed 2020-05-03 by SyntaxColoring